LumaTeam/Luma3DS #1240: Commit crashes to black screen when launching Super Smash Bros.

Searinox, 2019-04-06:
**System model:**

New 3DS XL

11.9.0-42E

**Entrypoint (How/what you're using to boot Luma3DS):**

B9S v1.3

**Luma3DS version:**

https://github.com/AuroraWright/Luma3DS/commit/70757e564d3cf9c371a60ee9e4bdd605d12a4bb6 (earlier are also possible, not sure when exactly this broke)

**Luma3DS configuration/options:**

Default EmuNAND: N/A

Screen brightness: ( 4 )

Splash: ( 1 )

PIN lock: ( Off )

New 3DS CPU: ( Off )

Autoboot EmuNAND: N/A

Use EmuNAND FIRM if booting with R: N/A

Enable loading external FIRMs and modules: ( False )

Enable game patching: ( True )

Show NAND or user string in System Settings: ( False )

Show GBA boot screen in patched AGB_FIRM: ( True )

Patch ARM9 access: N/A

Set developer UNITINFO: ( False )

Disable ARM11 exception handlers: ( False )

--

**Explanation of the issue:**

Aformentioned build fails to start Super Smash Bros.. Other extended memory titles were tried(Pokemon SM, MH4, Mario Maker) and they work fine. Once game is launched, the initial Nintendo 3DS splash screen is shown then transition to a black screen occurs, where it halts indefinetly. Interestingly enough, pressing A during that black screen will cause the console to restart.

Issue will occur regardless of whether just the title is installed or also update, external data, or save data presence. A clean reinstall of the CIA with delete title + ticket beforehand was attempted. The result did not change.

Issue does NOT occur on at LEAST luma build bd15f51a. It is unknown exactly at which commit the breaking took place.

**Steps to reproduce:**

1. Boot console on latest firmware with aformentioned build.

2. Launch Super Smash Bros.

**Dump file:**

N/A

BeastSE7, 2019-04-11:
I have looked over the entire Webs, I cannot find anything about this issue, now you have nailed exactly what is happening with me as well, on 2 N3dsXL units, Both go black screen on SSB and Detective Pikachu and Sonic Boom both f&i and shattered crystal!

Please can somebody help with this issue?
Both Units are 11.9.0.42E with B9S1.3 latest luma3ds

Searinox, 2019-04-11:
> I have looked over the entire Webs, I cannot find anything about this issue, now you have nailed exactly what is happening with me as well, on 2 N3dsXL units, Both go black screen on SSB and [...]

Hello, what version of Luma3DS are you using and if it's also a nightly, does it fix itself when using the commit I mentioned or even way back the official 9.1 release?

BeastSE7, 2019-04-12:
Ok yes, If I use the 9.1 release the system returns to normal and will play all games without issue.
Originally I used the latest 179-luma3ds-e57b546 because GodMode9 would show the touchscreen so I couldn't use any menus to finalise the install, I then installed FastBoot1.2 then godmode9 run without issue but the Console wouldn't play any extended memory games by the looks of things (SSB) then flashing 9.1 release into it, godmode9 has now touch screen again BUT I can play everything so I don't care now if GM9 doesn't work

MasterGamingYT, 2019-04-12:
Try with stable release, also people reported that luma hourlies brick consoles, according to luma updater

fireworksordie, 2019-04-15:
i can confirm this as well using last night's build for (commit e11cc09). detective pikachu opens and sits at black screen but works again when i revert to release 9.1. if i have some time tomorrow i'll try going back through nightlies to see if i can help pinpoint the start of the issue.

Searinox, 2019-04-15:
I went through the commits. The last commit that works and is available as binary from Kartik is 74bbfb9 . Commit e57abad is not available as compiled. The commit after that - 1041d7c - which is available as binary - breaks Smash Bros.. Starting with that commit and all commits after that, SSB no longer works.

TuxSH, 2019-04-16:
That's annoying. Which titles break, aide from SSB and Detective Pikachu? Does it happen on old3ds as well?

Searinox, 2019-04-16:
I do not have an o3ds to test with so waiting for other people to chime in.

Searinox, 2019-04-16:
I can now confirm this also happens on an O3DS. Same behaviour - black screen indefinetly, A restarts the console.

profi200, 2019-04-16:
Any exception dump on the SD card? It sounds like a crash with blackscreen.

TuxSH, 2019-04-16:
@profi200 or more like an errdisp, but with the screen not being initialized.

I can reproduce

Searinox, 2019-04-16:
Looked but couldn't find one. No dumps folder created even.

TuxSH, 2019-04-16:
luma/errdisp.txt (latest commit makes errdisp reimpl always log to a file), it's not a crash dump.

TuxSH, 2019-04-16:
It's a pm bug causing an error in kernel:

```
cpp
    if ( !KResourceLimit::GetMaxResource(processResLimit, RESLIMIT_CPUTIME) )
    {
      if ( owner->kernelReleaseVersion > 0x21Eu )// 2.1
        return 0xE0A01839;
```

I probably forgot to set reslimit properly, I'll have a more thorough look later.

TuxSH, 2019-04-17:
Latest commit should have fixed it
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/3087364/boot.zip)

fireworksordie, 2019-04-17:
can confirm your ZIP works (on my new 3ds). detective pikachu boots when it did not before

closed 2019-04-17 by Searinox

TheRozitaKnight, 2019-09-28:
I'm having the same issue on the latest release (v10.0.1): black screen after logo and A restarts console. I'm using an O3DSXL on 11.10.0-43U.

> Error type: generic
> Process ID: 40
> Process name: STORYOFS
> Process title ID: 0x0004000000142100
> Address: 0x0011dd80
> Error code: 0xf96047a2

grepo11, 2019-10-01:
running the latest release on a 2ds, I also want to point out that any button press restarts the console. the game i'm seeing this error for is omega ruby, errdisp:

Error type:       generic
Process ID:       38
Process name:     sango-1
Process title ID: 0x000400000011c400
Address:          0x001181e8
Error code:       0xf96047a2

torchicx19, 2023-01-09:
> Confirming SSB is working again with provided binary.

How did you did it? I have the same problem, please explain me how to fix it!