LumaTeam/Luma3DS #1241: Youtube

THEGUY3ds, 2019-04-06:
**System model:**

New 3ds xl

**SysNAND version (+emu/redNAND version if applicable):**

Sysnand 11.9.0-42E

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9strap 1.3

**Luma3DS version:**

9.1

**Luma3DS configuration/options:**

Default EmuNAND: N/A

Screen brightness: (4)

Splash: ( 1)

PIN lock: ( Off)

New 3DS CPU: ( Off)

Autoboot EmuNAND: (Off)

Use EmuNAND FIRM if booting with R: (Off )

Enable loading external FIRMs and modules: (Off)

Enable game patching: (True )

Show NAND or user string in System Settings: (True)

Show GBA boot screen in patched AGB_FIRM: (True)

Patch ARM9 access: (False)

Set developer UNITINFO: (False)

Disable ARM11 exception handlers: (False)

--

**Explanation of the issue:**
After the youtube app boots then I try to type something after 1 min of trying to enter the search it crashes the 3ds. Youtube used to run fine like it should with no crashes also a discord friend of mine also had the same problem and he was running latest luma.

**Steps to reproduce:**

1.Boot console on latest firmware with aformentioned build.

2.Boot Eur youtube app.

**Dump file:**
[crash_dump_00000003.zip](https://github.com/AuroraWright/Luma3DS/files/3051012/crash_dump_00000003.zip)

MasterGamingYT, 2019-04-07:
YouTube on 3ds is known to be buggy, so it's the app's fault, not luma's

Alejo8989, 2019-04-07:
The error is not that YouTube is buggy, because YouTube 3ds never crashed before i installed Luma and all stuff, anyway the error says: Arm11 error, Debug Event, Prefetch abort. I think that's not error of YouTube.

urherenow, 2019-04-07:
Oh, really? So you think Luma does prefetching?

MasterGamingYT, 2019-04-07:
Umm......Pretend I didn't say anything o_o"

Alejo8989, 2019-04-08:
Well, i don't think YouTube 3ds has this type of screens:

[image]

DMSalesman, 2019-04-08:
Those are Luma's exception handlers at work. Their purpose is to show what was going on at the time something crashed.

MasterGamingYT, 2019-04-08:
Also is it me or do I keep getting exceptions when I close youtube while a video is playing?

matthewjp, 2019-04-12:
I have the same issue ,ever since they updated the layout it crashes

matthewjp, 2019-04-12:
Like a while back many a week or 2 ago or somewhere between

aspargas2, 2019-04-12:
we could test for sure if this is a luma issue or a youtube app issue by booting native.firm from gm9 and see if the error still occurs

david4444, 2019-04-13:
This is the YouTube's app issue lol
Luma3DSs exceptions can be seen when your system crashes. That does not mean that Luma is the cause for it. Try booting YouTube with other CFWs or try disabling the crash handler. It will still crash, without the exception though. If an app crashes, and it has Lumas crash handler, it does not mean it is Lumas fault. for the case of YouTube, the app is outdated and buggy, I remember even before I hacked my 3DS it crashed a lot.

closed 2019-04-14 by TuxSH