LumaTeam/Luma3DS #1242: Fifa 15 crashes in a specific point of the playthrough.

ducreyi, 2019-04-14:
**System model:**

2DS

11.9.0-42E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

 Boot9Strap

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (off )

PIN lock: (off )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

While playing FIFA 15, after a specific match in the playthrough, the game crashes and gives the crash dump i submitted. This is repeatable, and only happens after that particular game, which is also needed to progress since you can't skip matches in this game (the bug happens even if you forfeit). 
I've already tried formatting the SD card (i checked with h2testw too) and reinstalling the game, restoring the extdata file, but the problem persists. From what i understand the game is rather rare/unplayed so it could definitely be the first time this happens.
It's not really a problem since i dont care that much about the game, but i'm curious about what could cause this.

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/3077741/crash_dump_00000000.zip)

TuxSH, 2019-04-14:
That's weird, it's a NULL pointer dereference.

ducreyi, 2019-04-15:
> 
> 
> That's weird, it's a NULL pointer dereference.

What would that imply? Could it be a corrupted extdata? I have no way to test it easily since it's in the middle of the playthrough, so if i just remove it it's gonna take a while to reach the same point.

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai