LumaTeam/Luma3DS #1244: Beug 11.9 Luma

StigiTT, 2019-04-15:
Hello I hacked my 3ds in 11.9 with b9s 1.3 and luma 9.1 but I have a lot of bugs with hombrew like for example fbi that freezes at launch 1 time out of 5 or hombrew that crashes me there with an error message Like the app had an error the console will restart

That's all I have for a bellow.

urherenow, 2019-04-15:
Did you install hblauncher_loader or are you launching the homebrew launcher with steelhax every time?

StigiTT, 2019-04-15:
steelhax doesn't work anymore after installing lima and yes I have hbluncher loader on hbl it tells me rosalina so yes I launch rosalina

urherenow, 2019-04-15:
Punctuation is important. I can’t tell from your comment if you’re using hblauncher_loader, or if you’re using Rosalina. Really, you might be helping other people out if you could do several tests launching hb via hblauncher_loader, then doing the same using Rosalina, and report if one gives you this issue more than the other (or not...).

StigiTT, 2019-04-15:
with rosalina I have no problem but with hbluncher loader its freeze once out of 2 times at launch

and fbi that appears when I launch it from the home menu 1 time out of 10 freeze and luma updater that once out of 5 freeze when it is necessary to check the update check it freezes and it leaves and it tells me an error has occurred the console will restart

urherenow, 2019-04-15:
Those are odd results. Especially loading FBI from the home menu. Have you checked your memory card for errors using H2testw? And is it formatted with 32k or 64k clusters?

StigiTT, 2019-04-15:
so h2testw told me that I have no error and my card will be and in fat32 32 cluster and for the content look at the picture
https://imgshare.io/image/ZCw8L
My Sd Card It’s Good

TuxSH, 2019-04-16:
This is not an assistance channel. Please go to https://discord.gg/C29hYvh

closed 2019-04-16 by TuxSH