LumaTeam/Luma3DS #1247: [bug] Newer commits cause console to be stuck in sleep if connected to wifi with manual DNS set

Searinox, 2019-04-16:
Right, so this is a very specific issue that took me about a month to pin down...

**System model:**

New 3DS XL

11.9.0-42E

**Entrypoint (How/what you're using to boot Luma3DS):**

B9S v1.3

**Luma3DS version:**

The exact version where this starts is yet unknown, but it is at least bd15f51a .

**Luma3DS configuration/options:**

Default EmuNAND: N/A

Screen brightness: ( 4 )

Splash: ( 1 )

PIN lock: ( Off )

New 3DS CPU: ( Off )

Autoboot EmuNAND: N/A

Use EmuNAND FIRM if booting with R: N/A

Enable loading external FIRMs and modules: ( False )

Enable game patching: ( True )

Show NAND or user string in System Settings: ( False )

Show GBA boot screen in patched AGB_FIRM: ( True )

Patch ARM9 access: N/A

Set developer UNITINFO: ( False )

Disable ARM11 exception handlers: ( False )

--

**Explanation of the issue:**

If a wifi access point is set and currently connected and the DNS is set to a custom value, leaving the console in sleep mode connected for about ~5 minutes results in the console failing to wake up from sleep mode. It is not just a black screen, the power led also continues to dim as if in sleep mode. The Luma3DS menu cannot be called. Static or dynamic IP settings do not matter. Manual DNS is enough for this to happen. Although the issue may not happen 100% of the time, the probability is very high(~80%).

Issue does NOT occur on commits 74bbfb9 and older.
Issue does NOT occur if the DNS is set to automatic.

**Steps to reproduce:**

1. Have a wifi access point available for the 3DS to connect to. DHCP needs to be available and the console needs to be able to auto-obtain both IP and DNS.
2. Have Luma3DS at least commit bd15f51a set up on the console.
3. Boot the console and launch the Configuration app.
4. Go to the wifi settings and set up connection to the wifi access point. However, under DNS settings, manually set the DNS. An example would be 8.8.4.4(Google DNS) and 208.67.220.220(OpenDNS) for the primary and secondary settings. Note, it does not matter if the setting matches what the router or ISP is providing, just that it is custom and not auto.
5. Verify that the access point works correctly and quit the Settings app.
6. Once booting back into the menu, check that an internet connection is acquired.
7. Close the lid on the console and wait for about 5 minutes. Note that the console correctly remain connected with the yellow led being lit with occasional flashing.
8. Open the console lid.

**Dump file:**

N/A

Searinox, 2019-04-16:
Fixed initial commit numbers, sorry.

WinterMute, 2019-04-16:
I can't reproduce this on New 3DS (non XL) with same firmware, b9s and luma from git HEAD.

M1CR0H4CK3R, 2019-07-09:
I have had this issue happen once WITHOUT manual DNS while playing MM3D via cartridge. @Searinox, do you have a cartridge inserted? If so, can you test this with it removed?

closed 2020-04-15 by TuxSH