LumaTeam/Luma3DS #1248: Unable to build master

RiderExMachina, 2019-04-17:
**Luma3DS version:**

Started after 8345f31c (I was able to build it).

**Explanation of the issue:**

Running `make` fails to make a build on the same machine I built 8345f31c on. I have makerom and firmtool in my path, as well as libctru and devkitARM. Running `# pacman -Syu` indicates there are no updates.

I'm on Fedora 29 with pacman setup for devkit development.

The error is as follows:
```
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function 'fsldrPatchPermissions':
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:46:11: error: implicit declaration of function 'FSREG_Register' [-Werror=implicit-function-declaration]
     res = FSREG_Register(pid, 0xFFFF000000000000LL, &info, &storageInfo);
           ^~~~~~~~~~~~~~
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function 'loader_GetProgramInfo':
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:230:11: error: implicit declaration of function 'FSREG_GetProgramInfo'; did you mean 'LOADER_GetProgramInfo'? [-Werror=implicit-function-declaration]
     res = FSREG_GetProgramInfo(exheader, 1, prog_handle);
           ^~~~~~~~~~~~~~~~~~~~
           LOADER_GetProgramInfo
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:234:11: error: implicit declaration of function 'FSREG_CheckHostLoadId' [-Werror=implicit-function-declaration]
     res = FSREG_CheckHostLoadId(prog_handle);
           ^~~~~~~~~~~~~~~~~~~~~
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:239:13: error: implicit declaration of function 'PXIPM_GetProgramInfo'; did you mean 'loader_GetProgramInfo'? [-Werror=implicit-function-declaration]
       res = PXIPM_GetProgramInfo(exheader, prog_handle);
             ^~~~~~~~~~~~~~~~~~~~
             loader_GetProgramInfo
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function 'loader_RegisterProgram':
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:421:13: error: implicit declaration of function 'PXIPM_RegisterProgram'; did you mean 'loader_RegisterProgram'? [-Werror=implicit-function-declaration]
       res = PXIPM_RegisterProgram(prog_handle, title, update);
             ^~~~~~~~~~~~~~~~~~~~~
             loader_RegisterProgram
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:443:9: error: implicit declaration of function 'FSREG_LoadProgram'; did you mean 'LOADER_LoadProcess'? [-Werror=implicit-function-declaration]
   res = FSREG_LoadProgram(prog_handle, title);
         ^~~~~~~~~~~~~~~~~
         LOADER_LoadProcess
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function 'loader_UnregisterProgram':
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:466:12: error: implicit declaration of function 'FSREG_UnloadProgram' [-Werror=implicit-function-declaration]
     return FSREG_UnloadProgram(prog_handle);
            ^~~~~~~~~~~~~~~~~~~
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:474:14: error: implicit declaration of function 'PXIPM_UnregisterProgram'; did you mean 'loader_UnregisterProgram'? [-Werror=implicit-function-declaration]
       return PXIPM_UnregisterProgram(prog_handle);
              ^~~~~~~~~~~~~~~~~~~~~~~
              loader_UnregisterProgram
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function '__appInit':
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:588:3: error: implicit declaration of function 'fsRegInit'; did you mean 'fsInit'? [-Werror=implicit-function-declaration]
   fsRegInit();
   ^~~~~~~~~
   fsInit
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:599:3: error: implicit declaration of function 'pxiPmInit'; did you mean 'srvPmInit'? [-Werror=implicit-function-declaration]
   pxiPmInit();
   ^~~~~~~~~
   srvPmInit
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function '__appExit':
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:605:3: error: implicit declaration of function 'pxiPmExit'; did you mean 'srvPmExit'? [-Werror=implicit-function-declaration]
   pxiPmExit();
   ^~~~~~~~~
   srvPmExit
/home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:608:3: error: implicit declaration of function 'fsRegExit'; did you mean 'fsExit'? [-Werror=implicit-function-declaration]
   fsRegExit();
   ^~~~~~~~~
   fsExit
```

I'm not sure if I need another devkit library installed or what's going on, but I have tried resetting the directory and just redownloading the build, but the issue persists.

Oreo639, 2019-04-17:
Luma depens on functionality that isn't in the latest release of libctru yet.

RiderExMachina, 2019-04-17:
@Oreo639 That makes sense. So I just need to wait until libctru updates and it should build successfully. Thanks!

closed 2019-04-17 by RiderExMachina

james-d-elliott, 2019-04-17:
You can just clone and build libctru as well. https://github.com/smealum/ctrulib

DEVKITPRO env must be set, and just do make install.

Shadow322, 2019-04-17:
> 
> 
> **Luma3DS version:**
> 
> Started after [8345f31](https://github.com/AuroraWright/Luma3DS/commit/8345f31c2a120a06d35e32bf85b5379613945968) (I was able to build it).
> 
> **Explanation of the issue:**
> 
> Running `make` fails to make a build on the same machine I built [8345f31](https://github.com/AuroraWright/Luma3DS/commit/8345f31c2a120a06d35e32bf85b5379613945968) on. I have makerom and firmtool in my path, as well as libctru and devkitARM. Running `# pacman -Syu` indicates there are no updates.
> 
> I'm on Fedora 29 with pacman setup for devkit development.
> 
> The error is as follows:
> 
> ```
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function 'fsldrPatchPermissions':
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:46:11: error: implicit declaration of function 'FSREG_Register' [-Werror=implicit-function-declaration]
>      res = FSREG_Register(pid, 0xFFFF000000000000LL, &info, &storageInfo);
>            ^~~~~~~~~~~~~~
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function 'loader_GetProgramInfo':
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:230:11: error: implicit declaration of function 'FSREG_GetProgramInfo'; did you mean 'LOADER_GetProgramInfo'? [-Werror=implicit-function-declaration]
>      res = FSREG_GetProgramInfo(exheader, 1, prog_handle);
>            ^~~~~~~~~~~~~~~~~~~~
>            LOADER_GetProgramInfo
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:234:11: error: implicit declaration of function 'FSREG_CheckHostLoadId' [-Werror=implicit-function-declaration]
>      res = FSREG_CheckHostLoadId(prog_handle);
>            ^~~~~~~~~~~~~~~~~~~~~
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:239:13: error: implicit declaration of function 'PXIPM_GetProgramInfo'; did you mean 'loader_GetProgramInfo'? [-Werror=implicit-function-declaration]
>        res = PXIPM_GetProgramInfo(exheader, prog_handle);
>              ^~~~~~~~~~~~~~~~~~~~
>              loader_GetProgramInfo
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function 'loader_RegisterProgram':
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:421:13: error: implicit declaration of function 'PXIPM_RegisterProgram'; did you mean 'loader_RegisterProgram'? [-Werror=implicit-function-declaration]
>        res = PXIPM_RegisterProgram(prog_handle, title, update);
>              ^~~~~~~~~~~~~~~~~~~~~
>              loader_RegisterProgram
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:443:9: error: implicit declaration of function 'FSREG_LoadProgram'; did you mean 'LOADER_LoadProcess'? [-Werror=implicit-function-declaration]
>    res = FSREG_LoadProgram(prog_handle, title);
>          ^~~~~~~~~~~~~~~~~
>          LOADER_LoadProcess
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function 'loader_UnregisterProgram':
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:466:12: error: implicit declaration of function 'FSREG_UnloadProgram' [-Werror=implicit-function-declaration]
>      return FSREG_UnloadProgram(prog_handle);
>             ^~~~~~~~~~~~~~~~~~~
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:474:14: error: implicit declaration of function 'PXIPM_UnregisterProgram'; did you mean 'loader_UnregisterProgram'? [-Werror=implicit-function-declaration]
>        return PXIPM_UnregisterProgram(prog_handle);
>               ^~~~~~~~~~~~~~~~~~~~~~~
>               loader_UnregisterProgram
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function '__appInit':
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:588:3: error: implicit declaration of function 'fsRegInit'; did you mean 'fsInit'? [-Werror=implicit-function-declaration]
>    fsRegInit();
>    ^~~~~~~~~
>    fsInit
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:599:3: error: implicit declaration of function 'pxiPmInit'; did you mean 'srvPmInit'? [-Werror=implicit-function-declaration]
>    pxiPmInit();
>    ^~~~~~~~~
>    srvPmInit
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c: In function '__appExit':
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:605:3: error: implicit declaration of function 'pxiPmExit'; did you mean 'srvPmExit'? [-Werror=implicit-function-declaration]
>    pxiPmExit();
>    ^~~~~~~~~
>    srvPmExit
> /home/rider/git/Luma3DS/sysmodules/loader/source/loader.c:608:3: error: implicit declaration of function 'fsRegExit'; did you mean 'fsExit'? [-Werror=implicit-function-declaration]
>    fsRegExit();
>    ^~~~~~~~~
>    fsExit
> ```
> 
> I'm not sure if I need another devkit library installed or what's going on, but I have tried resetting the directory and just redownloading the build, but the issue persists.

https://github.com/hax0kartik/luma-hourlies/releases

urherenow, 2019-04-18:
This isn’t a help forum, and you know this. Please post in temp or discord if it’s not a bug.

With 3ds projects, you will most often run into 2 issues.

1) on rare occasion, updating with git doesn’t quite work. In this case, delete your local repo and clone —recursive again.

2) the repo is using newer ctrulib functions, as you suspected. In this case, you need to manually clone, build, and install ctrulib. If in a ubuntu vm or similar, use “sudo -E make install”

RiderExMachina, 2019-04-18:
@urherenow The issue is resolved, there was no need for you to reply. Thank you for your concern.