LumaTeam/Luma3DS #1249: [BUG] Rhythm Heaven Megamix crashes with ARM11 Exception Occurred

AaronVey22, 2019-04-18:
System model: Old 3DS XL

SysNAND version: 11.9.0-42U (SysNAND)

Entrypoint: fastboot3ds 1.2

Luma3DS version: 9.1
(seriously where is 9.2 it's been forever)

Luma3DS configuration/options:

Default EmuNAND: ( )
Screen brightness: (4)
Splash: ( )
PIN lock: ( )
Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: (X)
Enable game patching: (X)
Show NAND or user string in System Settings: (X)
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: (It's not there)
Set developer UNITINFO: ( )
Disable ARM11 exception handlers: ( )

Explanation: The Game (Rhythm Heaven Megamix (USA)) Crashes with "ARM11 Exception Occurred"

Steps to reproduce:

1. Launch Rhythm Heaven Megamix

2. Cry because it crashed

Dump file:
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/3095584/crash_dump_00000000.zip)

AaronVey22, 2019-04-18:
Note that this also happens with boot9strap

TuxSH, 2019-04-18:
0xD9004587 in register dump => you have a corrupt title installation. Not a Luma issue

> Note that this also happens with boot9strap
of course, this is totally unrelated. fb3ds has really no advantage over b9s if you're not actively using its features... (and latest Luma3DS commit doesn't cause any issue in gm9)

> (seriously where is 9.2 it's been forever)

soon. I still need to fix some critical bugs and stuff like that.

closed 2019-04-18 by TuxSH