LumaTeam/Luma3DS #1250: MK7 Plugin causing crashes

TopLelXD, 2019-04-26:
**System model:**

Old 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

Whatever the latest NTSC-U version is

**Entrypoint (How/what you're using to boot Luma3DS):**

The latest version of Boot9Strap

**Luma3DS version:**

v9.1-4b341e03 (Nanquitas's plugin version)

**Luma3DS configuration/options:**

Default EmuNAND: ?

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

--

Autoboot EmuNAND: ?

Use EmuNAND FIRM if booting with R: ?

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

MK7 Live Replay code (found here: https://gbatemp.net/threads/release-mario-kart-7-online-ntr-plugin.460631/) gives an error and forces me to reboot my 3DS

**Steps to reproduce:**

1. Enable the plugin

2. Boot Mario Kart 7

3: Turn on the Live Replay Code - Go into the plugin by pressing select -> Menu & Other Codes -> Check "Live Time Trial Ghost Replay"

4: Select character, vehicle combo and ghost you want to view

5: Restart the time trial (required to view the live replay)

6: The 3DS crashes

**Dump file:**

Never got one

david4444, 2019-04-27:
Since you said it is "Nanquitas's plugin version" I'm pretty sure this isn't an issue with Luma3DS.
If I didn't know you can post an issue about this version then sorry.

profi200, 2019-04-27:
It's not official Luma and the official version doesn't even have plugin support so it can only be the inofficial version or a buggy plugin causing it. Open an issue there.

closed 2019-04-27 by TuxSH