LumaTeam/Luma3DS #1252: Locale switch not working for some VC games

subworx, 2019-04-29:
**System model:**

N2DSXL

**SysNAND version:**

11.9.0-42E

**Entrypoint (How/what you're using to boot Luma3DS):**

SIGHAX

**Luma3DS version:**

9.1-https://github.com/AuroraWright/Luma3DS/commit/8b56098d64754a7452f6defb437fbd978e3950d2 nightly (latest at time of reporting) from https://github.com/hax0kartik/luma-hourlies/releases/tag/199-luma3ds-8b56098

**Luma3DS configuration/options:**

Default EmuNAND: (1)
Screen brightness: (4)
Splash: (Off)
PIN lock: (Off)
New 3DS CPU: (Off)

Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: (x)
Show NAND or user string in System Settings: (x)
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )
Disable ARM11 exception handlers: ( )

**Explanation of the issue:**

System is EUR, all installed games US.
Some crashed, creating `sd:/luma/titles/<titleID>/locale.txt` (`USA EN`) fixes this for all of them.
Recently installed lots of US or region free Virtual Console games, did the locale.txt thing for all of them to be safe.
Some still keep crashing.
All of them have 000400000Fxxxxxx in the titleID, which makes me think Luma has issues with properly recognizing that ID, but why would it?
Also proven wrong as I have 13 titles with 000400000Fxxxxxx but only 8 crash.
One NES game is affected, and 7 SNES games. 
Other NES/SNES and all GB/GBC/GBA VC games I tested work fine - all of them do NOT have the 'F' in their titleID, e.g. Mega Man 1-6 USA (NES, titleID 00040000000xxxxx) work, MM7 USA (000400000F702800) doesn't.

000400000F707D00 - VC Castlevania Dracula X (US) - Works
000400000F700100 - Xenoblade Chronicles 3D (US) - Works - a real 3DS title
000400000F700800 - The Binding of Isaac: Rebirth (US) - Works - another real 3DS title
000400000F701000 - VC Zelda: Link to the Past (RF) - Works
000400000F701600 - VC Earthbound (USA) - Works

000400000F700E00 - VC Super Mario World (RF) - Crash
000400000F701200 - VC Super Metroid (RF) - Crash
000400000F701900 - VC F-Zero (RF) - Crash
000400000F702200 - VC Pilotwings (RF) - Crash
000400000F702600 - VC Mega Man X (RF) - Crash
000400000F702800 - VC Mega Man 7 (US) - Crash
000400000F703500 - VC Super Castlevania IV (US) - Crash
000400000F706800 - VC Demon's Crest (US) - Crash

**Steps to reproduce:**

1. On an EUR device, install one of the games on the list marked with "Crash"
2. Create `sd:/luma/titles/<titleID>/locale.txt` according to the installed game and fill it with `USA EN`
3. Launch the game
4. An exception occurred, see attached dump

**Dump file:**
[crash_dump_00000003.zip](https://github.com/AuroraWright/Luma3DS/files/3128319/crash_dump_00000003.zip)

TuxSH, 2019-05-01:
Are the errors all similar? The error code I see in the dump is 0xd9004587, indicating the title is corrupt/you installed a corrupt title.

closed 2019-05-01 by TuxSH

subworx, 2019-05-02:
Seeds were missing (thought I placed them into the correct folder before installing), and I feel really stupid now. Sorry for the trouble.