LumaTeam/Luma3DS #1255: Crash when trying to open Homebrew Launcher

DarthVitrial, 2019-05-09:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.9 

**Entrypoint (How/what you're using to boot Luma3DS):**

FastBoot 3DS 1.2 Stable

**Luma3DS version:**
https://github.com/AuroraWright/Luma3DS/commit/2c8a1eb5c644ed29289601c4c9a121e0486d47fb

**Luma3DS configuration/options:**

Brightness: 4x

New 3DS CPU: Clock + L2

Enable game patching: (yes )

Show NAND or user string in System Settings: (yes )

Show GBA boot screen in patched AGB_FIRM: (no

 )

Patch ARM9 access: (no )

Set developer UNITINFO: ( no)

Disable ARM11 exception handlers: (no )

--

Trying to open the Homebrew Launcher (https://github.com/mariohackandglitch/homebrew_launcher_dummy/releases/tag/v1.0) causes an Undefined Instruction crash.

**Steps to reproduce:**

1. Install Homebrew Launcher

2. Open Homebrew Launcher

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/3163643/crash_dump_00000000.zip)

urherenow, 2019-05-09:
Then why not make the issue there? Not a Luma issue. I’ve used hblauncher_loader for years, and don’t have an issue.

DarthVitrial, 2019-05-09:
Because I thought it was caused by some change in the newer Luma builds, since it worked for me when I tried it on older Luma.

closed 2019-05-30 by TuxSH