LumaTeam/Luma3DS #1257: I have a problem

Axelpineda, 2019-05-26:
since I pirated my 3ds and had not so many good problems, let them explained: first I can install themes I can also install them I can also play the games but ... here is the problem because I can not play all the games :( this is well know because I can play fire emblem and zelda and mario and many more but the question is that the games that I want most are only two Pokemon ultra Sun and Super smash bros I can not play them when I install them the image of the 3ds and everything ok but then the black screen is left and every now and then I get an error of ARM11 and the truth is that it is not because I have the last verification of luma, gm9 and B9S and since they are the only games (tested) that they give me that error and that everything else works, I do not know what to expect, I hope someone can help me and explain me a little because I do not understand much about those compilation and programming things.

fincs, 2019-05-26:
Piracy is not supported.

closed 2019-05-26 by fincs

Axelpineda, 2019-05-27:
osea que

fincs, 2019-05-27:
Que no, que no damos ningún tipo de soporte alguno a la piratería. En tu post inicial has admitido que estabas instalando juegos pirateados, con lo que no te podemos ofrecer ninguna ayuda. Solo nos ponemos a investigar si hay errores o fallos con juegos adquiridos legítimamente, como los de la eShop.

> Nope, we don't support piracy at all. In the OP you admitted to installing pirated games, so we cannot offer you any help. We only troubleshoot issues or bugs in legally bought games (such as those from the eShop).