LumaTeam/Luma3DS #1258: PM Random Crash after some Game Starts

Bappsack, 2019-05-26:
**System model:**
N3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.9

**Entrypoint (How/what you're using to boot Luma3DS):**
BS9
**Luma3DS version:**
9.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (x)

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: (x)

--

**Explanation of the issue:**

pm process crash if u reopen a game multiple times, PID increases, after PID 50-100 reached it crash at some point.

**Steps to reproduce:**

1. Start a Game, close it, start it again.

**Dump file:**

Ammako, 2019-05-29:
What do you mean by "random"? How many times have you reproduced this, and how rarely does it happen? I'm up to a process ID of 234 and haven't been able to reproduce a crash (game patching disabled.)

It could also be important to specify which games are resulting in a crash.

Also, you really should read the warning when you chose to disable ARM11 exception handlers. Re-enable them, and then comment again, this time including the information that the exception handler gives you.

Bappsack, 2019-05-29:
Pokemon Ultra Sun, it happens not anytime but usually after you reopen the
game.

TuxSH, 2019-05-30:
Disable ARM11 exception handlers: (x)

Please enable this option, also is this a recent commit, if so, which one?

Bappsack, 2019-05-30:
it is disabled, not a recent commit, stable 9.1

i made a Game Restart script with Luma Input Redirection, Power Button(Short), HOME Button, Re-Enter Game. PID increases and at a random ID it crashes with the Luma Exception handler, says PM crashed.

not really sure if its on my side or Luma, should also mention i use NTR aswell for Memory Read/Writes wich are not used in the script at all, just NTR runs in background.

Sonlen1414, 2019-05-30:
NTR doesn't play nicely at all with game restarts, it is actually impressive it works past the first, test again without NTR and see if it keeps doing it.

SCOTT0852, 2019-05-30:
NTR is probably the issue here, it's really hacky and usually I can only start a single game correctly with the next one always hanging on the "Nintendo 3DS" splash screen.

Bappsack, 2019-05-30:
@SCOTT0852 not exactly true, the "hang" is caused by the Streaming Feature, i don't use that feature, without it works fine(mostly).

Sonlen1414, 2019-05-30:
"Mostly" being the keyword here, I would put money the issue is NTR and not Luma here.

Ammako, 2019-05-30:
> @SCOTT0852 not exactly true, the "hang" is caused by the Streaming Feature, i don't use that feature, without it works fine(mostly).

It seems you have information about the problem which you haven't brought up in your issue report. You should include as much information as you can when writing a report.

Bappsack, 2019-05-30:
its running rn, will post the crash dump after it crash

Bappsack, 2019-05-30:
looks like its cuz NTR, im on PID 250+ now, no crash yet, ill upload a crash dump aswell for the crash with ntr, maybe some1 knows why exactly.

[crash_dump_00000067.zip](https://github.com/AuroraWright/Luma3DS/files/3238578/crash_dump_00000067.zip)

TuxSH, 2019-06-12:
See https://twitter.com/TuxSH/status/1138556585536565248

closed 2019-06-12 by TuxSH