LumaTeam/Luma3DS #1260: GDB remote connect fails

ParzivalWolfram, 2019-06-02:
**System model:**

N3DS (RED model)

**SysNAND version (+emu/redNAND version if applicable):**

11.10.0-43U SysNAND, no EmuNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

boot9strap installed via ntrboot

**Luma3DS version:**

v9.1

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (2)

Splash: (Before, 3 seconds)

PIN lock: (8 digits)

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: (No)

Use EmuNAND FIRM if booting with R: (No)

Enable loading external FIRMs and modules: (No)

Enable game patching: (Yes)

Show NAND or user string in System Settings: (Yes)

Show GBA boot screen in patched AGB_FIRM: (Yes)

Patch ARM9 access: (Not available as an option)

Set developer UNITINFO: (No)

Disable ARM11 exception handlers: (No)

--

**Explanation of the issue:**
Rosalina debugger causes GDB to refuse to connect, due to:

`Remote "g" packet reply is too long: <large hex string purged>`

**Steps to reproduce:**

1. Start Rosalina debugger

2. Attach debugger to any process

3. Attempt connect with GDB

**Dump file:**

Not applicable, crash does not occur.

Oreo639, 2019-06-02:
Are you using gdb or `$DEVKITARM/bin/arm-none-eabi-gdb`.
I would recommend making an alias for executing that if you are on *nix.

ParzivalWolfram, 2019-06-03:
a GDB binary I cross-compiled for my phone.

piepie62, 2019-06-03:
Well, that's unlikely to work, then

ParzivalWolfram, 2019-06-03:
Works for other devices.

Oreo639, 2019-06-03:
The version dkp provides adds patches for the 3ds.

fincs, 2019-06-03:
Closing issue, user error determined as cause.

closed 2019-06-03 by fincs