LumaTeam/Luma3DS #1262: Game patching non-functional

Blockmaster139, 2019-06-05:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.10.0-43E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v9.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: 2(x)

Splash: Off(x)

PIN lock: Off(x)

New 3DS CPU: Clock+L2(x)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Patch files located in luma/titles/[titleid] have stopped being recognized, or the patch function has stopped working, causing them to not be loaded. Tested with an undub for Etrian Odyssey V, the game will load the default English dub voices. Undub files are in the correct location and installed properly, and the function was working when I played the game last year, and I have not edited or changed the undub files since. 

**Steps to reproduce:**

1. Add patch files to luma/titles/[titleid] folder

2. Load game

vibe, 2019-06-09:
I also have an .ips I am trying to apply to patch a specific module and no success.

TuxSH, 2019-06-12:
Is this using luma on sd card? Other people can't reproduce.

Blockmaster139, 2019-06-12:
Not sure what you mean exactly, but I was able to open the settings menu and Rosalina with the microSD removed

RocketRobz, 2019-06-12:
Luma is on the CTRNAND then.
Maybe that's the issue?

Blockmaster139, 2019-06-13:
I can't get it to boot from the SD card, CTRNAND always takes priority. Will I need to uninstall it from CTRNAND and how do you do that?

TuxSH, 2019-06-13:
boot9strap tries to boot boot.firm from SD card, then from NAND. Is your SD card working, do you have a proper setup, etc.?

Blockmaster139, 2019-06-13:
SD card definitely works, my games and themes load normally. Luma is installed to the SD exactly as instructions describe. My indication that it's running from CTRNAND is from the options menu when holding select

Ammako, 2019-06-17:
Can you confirm that your boot.firm on your SD card is not corrupted or invalid? Redownload, and make sure the checksum matches after copying it to your SD card. Assuming you're using Windows, you can check the checksums by shift + right clicking in the folder where boot.firm is located, clicking "Open command window here", and doing `certutil -hashfile boot.firm MD5`.

Do this for both the original file you just redownloaded, and the copy you made on your SD card, to confirm that it isn't corrupted, and see if it works then.

Blockmaster139, 2019-06-18:
Checksums match up, I tested right after transferring the file and after booting the 3DS once. Still no luck

TriforceOfPwnage, 2019-06-19:
I can confirm the same problem on mine. My boot.firm doesn't seem to be corrupted either

TriforceOfPwnage, 2019-06-30:
Still happens on latest official release for me

closed 2019-07-03 by TuxSH