LumaTeam/Luma3DS #1268: Blue light level reset when waking console from sleep mode

RocketRobz, 2019-06-29:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.10.0-43U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

10.0 stable

**Luma3DS configuration/options:**

Screen brightness: (1)

Splash: (Before)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: (X)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

After setting the blue light level in Rosalina menu, waking the console from sleep mode resets the level to 0.

**Steps to reproduce:**

1. Open Rosalina menu.

2. Set blue light level to 1-5.

3. Exit Rosalina menu.

4. Close the console's lid.

5. Open the console's lid.

TuxSH, 2019-06-29:
That's normal. This is because when there was something to detect sleep mode wakeup, sometimes the 3DS (and in particular 2DSes) failed to wake up, henceforth I decided to remove this aspect of the feature.

closed 2019-06-29 by TuxSH

AndrewHend, 2019-06-30:
I guess this is something that will never be changed, or looked into improving?  I too would really like the blue light filter setting to stick and be remembered, once set.  Every time I reboot/power up the system from being off, I have to set this again.  It would be really convenient if it could be remembered, for those who would permanently like to have strain reduced on their eyes, or for those that regularly play in darker lighting.

DMSalesman, 2019-07-01:
@AndrewHend @RocketRobz See #1222.

AndrewHend, 2019-07-01:
Thank you so much for linking me to the above.  I really do appreciate it and did not realize all the hard work that was already attempted, in being able to get the blue light filtering settings to stick, once set.  It seems like it was a real hassle and I was ignorant to that fact.  We shouldn't be complaining, considering this is done for free.  I guess it would be a nice surprise if this was figured out down the road, but for now I am understanding and accepting.

Thanks again so much for everyone's hard work on this matter and on Luma, in general.

jbmagination, 2020-03-30:
I guess a solution for this could be to turn it off when sleep mode is on, then to re-enable it when it is off? Blue light is the direct problem, sooo