LumaTeam/Luma3DS #1270: Luma3DS v10.0 (N3DS EU, 11.10)

sambarringr, 2019-06-29:
Hi,
Whenever I leave microSD Management I get this error:
[Error](https://imgur.com/a/LyOfOwc)
All files are in their right place on the SD card

Never happened before I put Luma3DS v10.0 on, i was on 9.1 before

NTR is not open.
MicroSD management is being closed normally.

Crash dump: 
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/3342315/crash_dump_00000000.zip)

Thanks!

sambarringr, 2019-06-29:
Forgot to add this:
[image]

sambarringr, 2019-06-29:
ugh it rotated on github sorry

TuxSH, 2019-06-29:
It's normal, it's just microSD crashing even w/o Luma3DS. Luma3DS just makes the silent crash not-so-silent anymore.

tldr just ignore it

closed 2019-06-29 by TuxSH

AndrewHend, 2019-06-30:
I am having the exact same issue and like the original poster, this was NEVER an issue before Luma 10.0.  With stock 3DS firmware, I have never once had a crash when leaving the MicroSD management and with Luma up to 9.1, this was never an issue either.  I was ALWAYS able to successfully return to the home screen, when leaving the MicroSD management.  Whatever changed in Luma 10 is worse than before, even if it is 'ok' by your standards.  I would rather have it silent again, so I am not forced to have to power down my system, every time I want to return to the home screen, after putting files onto my 3DS through MicroSD management.

RocketRobz, 2019-07-01:
Is it not possible to just disable the Luma exception screen just for the MicroSD management app, and then re-enable, once you're back in System Settings?