LumaTeam/Luma3DS #1272: Pokemon Ultra Moon cheats very unstable on v10

Level99OCR, 2019-06-30:
After updating to v10, enabling a small amount of the cheats for my device that used to work just fine result in a number of crashes at various points in Pokemon Ultra Moon. The two that I'm able to absolutely pinpoint as being a culprit are the Wild Pokemon Shiny toggle and the Hold Start and open Options to access PC. See attached dmp
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/3342797/crash_dump_00000000.zip)

.

piepie62, 2019-06-30:
Could you provide the actual text of the codes, along with your installed Ultra Sun version and the version the codes were made for?

piepie62, 2019-06-30:
Ultra Moon, rather

Level99OCR, 2019-06-30:
Cheats were from this file: https://github.com/FlagBrew/Sharkive/blob/2b75942800d80a48bedf6d1658f1efb4083362ab/db/00040000001B5100.txt

Retail digital download of Ultra Moon patched to v1.2 

Specifically, these codes as far as I can tell. I rolled back to v9.1 but can do various configuration testing to be sure if you'd like.

[(START+L)Wild Pokemon Shiny, (START+R)Disable v1.2]
605BBFFC E12FFF1E
005BBFF0 E5C40004
005BBFF4 EA000000
005BBFF8 E5C41006
005BBFFC E12FFF1E
003A8690 EB084E56
D0000000 00000000
DD000000 00000208
005BBFF4 E3A01001
D0000000 00000000
DD000000 00000108
005BBFF4 EA000000
D0000000 00000000

[(Hold START and Open OPTIONS menu)Access the PC from anywhere v1.2]
E05BBEE0 00000088
E92D400E E59F1074
E5912000 E3A03000
E5813000 E3A03C13
E2833093 E1520003
03A04000 11A04000
E8BD800E E350000E
EAF70DE3 E92D400F
E24DD010 E3A00000
E58D0000 E58D0004
E58D0008 E58D000C
E1A03000 E1A02000
E3A01C13 E2811093
E59F0018 E5801000
E59F0014 E5900000
EBF77D15 E3A00002
E28DD010 E8BD800F
00669000 006A59C4
0037DB28 EB08F8F7
0038B030 EB08C3AA
DD000000 00000008
005BBF10 1AF70DE3
D0000000 00000000

piepie62, 2019-06-30:
What's your 3DS system version?

Level99OCR, 2019-06-30:
11.10.0-43U

TuxSH, 2019-06-30:
This seems to be a kernel bug (null deref), blame Nintendo I guess.

TuxSH, 2019-06-30:
There's a race condition in the kernel between the moments when:

* CreateThread/Run attaches the KThread to a KDebugThread
* The thread runs or any scenario involving thread->debugThread->x
basically you have to pray this doesn't happen

Ammako, 2019-07-02:
I don't know if this is the same issue, but here goes:

Ultra Sun
Same Luma and system versions

[No Outlines v1.2]
0042FBA4 E320F000

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/3352148/crash_dump_00000000.zip)

jql6, 2019-08-03:
V1.5 Pokemon Y

Using this: https://github.com/FlagBrew/Sharkive/blob/master/db/0004000000055E00.txt

[(Hold R)Walkthrough Wall v1.5]
D3000000 00000000
0053ED50 E1A01000
0053ED54 E12FFF1E
0053ED58 E1A06000
0053ED5C E12FFF1E
DD000000 00000100
0053ED50 E3A01000
0053ED58 E3A06000
D0000000 00000000
580B5824 E1A01000
080B5824 EBF89149
D0000000 00000000
580B3A20 E1A06000
080B3A20 EBF898CC
D0000000 00000000

It doesn't work, and crashes afterwards.

ericjwg, 2019-09-17:
any solution for .cro file memory region modification?
like this:
[No Encounter, Hold START for Encounter]
5807E5B8 E3A00064
0807E5B0 E3A09000
DD000000 00000008
0807E5B0 E3A09064
D0000000 00000000
D0000000 00000000

the cheat displayed in red after enable it, probably cause luma can't access the offset.(same goes with nanquitas's ctrpf AR feature, its built-in memory editor doesnt have a region include 0x0807XXXX either)

AverageComet250, 2020-05-19:
I rarely get crashes on ultra moon

suppai, 2022-06-05:
Nintendo bug, not related to Luma3DS; resolving.

closed 2022-06-05 by suppai