LumaTeam/Luma3DS #1276: 3DS Luma Exception Error (Prefetch Abort)

ZombieBrine13092, 2019-07-01:


ZombieBrine13092, 2019-07-01:
Change the crash dump file extension to a DMP please: crash_dump_000000004.DMP
This is due to GitHub's file type limit..

[crash_dump_00000004.txt](https://github.com/AuroraWright/Luma3DS/files/3343675/crash_dump_00000004.txt)

[image]

Validusername16, 2019-07-01:
YouTube on the 3DS is abandoned and that is normal.

ZombieBrine13092, 2019-07-01:
If it's abandoned, why am I still able to connect to YouTube anyway? It seems to me as if they should've removed it from the eShop by now.

TurdPooCharger, 2019-07-01:
@ZombieBrine13092,

- [The 3ds YouTube app doesn’t work anymore, try to watch a video and it force closes, maybe due to the low RAM the 3ds has and the YouTube website maybe had an minor update](https://old.reddit.com/r/3DS/comments/bsg3hi/the_3ds_youtube_app_doesnt_work_anymore_try_to/)

- [YouTube app on Nintendo 3DS is seemingly broken, if you care](https://www.nintendoenthusiast.com/2019/05/25/youtube-app-on-nintendo-3ds-is-seemingly-broken-if-you-care/)

- [No, the YouTube for 3ds app does not work anymore and keeps crashing. Here's how you can still use YouTube on the 3ds.](https://old.reddit.com/r/3DS/comments/bsvbpl/no_the_youtube_for_3ds_app_does_not_work_anymore/)

- [3DS' YouTube app appears to be broken](https://www.gonintendo.com/stories/336110-3ds-youtube-app-appears-to-be-broken)

SuperLolMine, 2019-07-02:
That happened to me too even **before** soft modding my 3DS with Luma. It said "An error has occurred", something along those lines after going to the YouTube app. Luma shows that log info because "Disable ARM11 exception handlers" is disabled by default, if you enable it it shows exactly what would pop up on a non-modded 3DS, the generic "An error has occurred." You can try this for yourself, but i recommend disabling that option after testing so if a crash happens in the future you have some info and you can generate a crash dump, instead of that generic error message.
The 3DS's YouTube app is just a clunky web browser that connects to the mobile YouTube homepage. If you connect to Youtube with the internet browser it works 100 times better.

closed 2019-07-03 by TuxSH