LumaTeam/Luma3DS #1277: Crash upon exiting microSD Management from setting

whichnametouse, 2019-07-01:
**System model:** New 3DS

**SysNAND version (+emu/redNAND version if applicable):** 11.9.0-42U then 11.10.0-43U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap 1.3

**Luma3DS version:** v10.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (1)

Splash: (Off)

Splash duration: (1s)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

After updating to luma 10.0, I was still on 11.9.0-42U, I opened setting then microSD management, copied some files to the 3ds then pressed Close button waiting for it to disconnect then restart, crash happened. I tried updating to 11.10.0-43U, same crash and I have tried several times to make sure it is a persistent crash, not occasionally. I use the boot.firm from latest update as in after the hotfixes. I havent encountered this issue before 10.0, was doing pretty fine with it and use the management pretty often.

**Steps to reproduce:**

1. Go to microSD Management on system settings, and connect to network.

2. After that, press Close.

**Dump file:**

[crash_dump_00000007.zip](https://github.com/AuroraWright/Luma3DS/files/3343984/crash_dump_00000007.zip)

RocketRobz, 2019-07-01:
Duplicate of #1270

closed 2019-07-01 by TuxSH

AndrewHend, 2019-07-03:
Thanks so much for fixing this!  Luma 10.0.1 is working perfectly now and just as 9.1 was, in terms of MicroSD management no longer crashing, upon exit.

Happy guy here!