LumaTeam/Luma3DS #1279: Exception Error after using MicroSD Management

Pokemiah, 2019-07-01:
**System model:**
New 3DS XL
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.10.0-43U SysNAND
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Sighax
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
v10.0 stable
[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Shortly after I updated to Luma v10.0 I went to update some of my homebrew programs. Since I use the New 3DS XL I can't exactly get the microSD card without unscrewing the back so instead I use the microSD management option in the Data management menu in the System Settings. As soon as I exited out of the process the 3DS just crashes giving me an exception error under the pretense of prefetch abort (svcBreak). I didn't have this issue until after the recent update of Luma3DS. I can consistently recreate the issue so I believe it's a flaw in Luma. However, none of the files I transferred seem to have corrupted.

**Steps to reproduce:**

1. Go to System Settings.

2. Select Data Management.

3. Select MicroSD Management.

4. Exit MicroSD Management. (It seems to crash whether or not you do anything in there.)

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/3347102/crash_dump_00000000.zip)

SuperLolMine, 2019-07-02:
Yup, same thing happens to me on Luma 10.0. I restored my old Luma version (9.1) for now and on that older version it does not crash.

TuxSH, 2019-07-02:
Duplicate. Latest commit fixes it, trying to find layeredfs regression before doing a new hotfix release.

closed 2019-07-02 by TuxSH

AndrewHend, 2019-07-02:
Thanks so much for doing this guys, really appreciate all the hard work.  I look forward to this new hotfix.