LumaTeam/Luma3DS #1280: How to using LayeredFS patch after game updated?

wzhy90, 2019-07-03:
**System model:**

 New 3DS

**SysNAND version :**

11.10.0-43J SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v9.1/v10.0 stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: (L2 )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( x )

Show NAND or user string in System Settings: ( x )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
I own a game card, and this game has an fan-translate patch. I patch it and works well, but after I install the game update, some of the resources restore to default. my update cia is also fan-translated. I try to extract romfs from the update cia and merge it to the origin patch and put the updated code.bin and extheader.bin in /luma/titles/TID/, but it crash while I launch it.(I have uninstall the update cia before trying this)

**Steps to reproduce:**

1.

2.

**Dump file:**
[crash_dump_00000000.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/3355005/crash_dump_00000000.dmp.zip)

TuxSH, 2019-07-03:
There's a LFS bug that I'm trying to fix rn

closed 2019-07-03 by TuxSH

wzhy90, 2019-07-08:
@TuxSH after upgrade to 10.0.1, the game only load LayeredFS patch but the upgrade of the game didnt load.
Did upgrade package support by the game patching?

commit c913de1e (2019-07-03): Fix layeredfs, fix #1280 #1275 #1262 & others