LumaTeam/Luma3DS #1281: New Super Mario Bros 2 crashing with luma 10.0 and 10.0.1

meepingsnesroms, 2019-07-04:
The game is just playing normally and then it crashed with errDisp:
[image]
I never had this issue with luma 9.1 but also dident play the game much there so I may have just never triggered it, it did happen twice in 2 days though and I played < 2 hours each time.

This crash has happened twice now causing loss of save data both times.
First time was on 10.0, second was on 10.0.1, both give the same error.

Firmware 11.10.0-43U, New 2DS.

meepingsnesroms, 2019-07-04:
Also not using any cheats or custom modules, just luma and the game.

lucasaf04, 2019-07-04:
Same with super Mario maker

GlaZedBelmont, 2019-07-04:
Ok, so @lucasaf04 , are you sure its the exact same error screen and what were you doing right before it crashed?

GlaZedBelmont, 2019-07-04:
and can you also test on luma 9.1 please

fincs, 2019-07-04:
Is this a cartridge, or a digitally purchased game? If the latter, would it be possible for you to check your SD card for errors, or try a different SD card?

GlaZedBelmont, 2019-07-05:
this seems to be an issue related to luma 10 itself , errdisp crash screens will not show properly and appears fucked up.

lucasaf04, 2019-07-05:
Yes, same error. @GlaZedBelmont

GlaZedBelmont, 2019-07-05:
yeah, as I said, you two are actually getting normal errdisp errors if we call them so, but the errdisp error screens probably broke with luma 10+

brunocar, 2019-07-17:
im getting similar errors when booting up heroes of ruin, its also an errdisp

TuxSH, 2020-04-17:
Should probably be fixed by latest commits?

closed 2020-04-17 by TuxSH

CedarCoder1, 2024-08-21:
I still experience random crashes in the older versions and latest versions of luma3ds. In the latest releases the crash shuts the system down and you'll need to replace all the luma3ds files for the handheld to boot again.