LumaTeam/Luma3DS #1293: Heroes Of Ruin (EU/US) crash with luma 10.0.1

lkclip, 2019-07-17:
**System model:** Old 3DS

**SysNAND version (+emu/redNAND version if applicable):** 11.10.0-43E

**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap

**Luma3DS version:** 10.0.1

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (Off )

Splash duration: (1x )

PIN lock: (Off )

--

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Digital image of Heroes Of Ruin (US) crashes during the first loading screen, and (EU) crashes after street pass notification request. It seems like the crash happens on autosave, because once save is created on EU version (by loading external Luma3DS 9.1 firm) the game launches fine. Issue confirmed by me and another redditor

**Steps to reproduce:**

1. Launch the title

**Dump file:** Error message doesn't show file path
[image]

brunocar, 2019-07-17:
Can confirm that I get the same error on the same game on a New 3DS XL with the same Luma version and similar settings.

hunttung, 2019-07-17:
I too, get the same error on 10.1

xerxesmanflayer, 2019-08-16:
I'm experiencing the same error on an old 2DS and a new 3DS, both running the same version of Luma

meepingsnesroms, 2019-08-18:
Can someone test this with "Disable ARM11 exception handlers" on, I have crashing in New Super Mario Bros 2 but its not reliable like it is in this game.

brunocar, 2019-09-10:
> 
> 
> Can someone test this with "Disable ARM11 exception handlers" on, I have crashing in New Super Mario Bros 2 but its not reliable like it is in this game.

already tried it, doesnt change anything

brunocar, 2019-09-19:
Any updates on this?

xerxesmanflayer, 2019-09-19:
I'm also curious if anyone is actively working on this/ has made any progress yet.
I'm very interested in playing this with my girlfriend, and have had no luck as of yet.

brunocar, 2019-10-05:
Right now the only workaround is simply downgrading to the previous luma version

xerxesmanflayer, 2019-10-20:
I haven't had any issues with Smash Bros, but I haven't tried NSMB2 or MHG yet.

Does anyone have any idea what the cause of the issue might be?

brunocar, 2019-10-20:
As already stated in another related issue, it seems that someone messed up and mixed up the check calls so that certain positive responses are instead replaced with the errdisp response, which is there to report errors.

xerxesmanflayer, 2019-10-20:
Oooh, ok. Thanks for the information.

Oreo639, 2019-10-20:
> As already stated in another related issue, it seems that someone messed up and mixed up the check calls so that certain positive responses are instead replaced with the errdisp response, which is there to report errors.

The information returned by errdisp on 10.0.1 is all garbage.
It didn't actually return 'sucess'.

> Can someone test this with  "Disable ARM11 exception handlers" on, I have crashing in New Super Mario Bros 2 but its not reliable like it is in this game.

 "Disable ARM11 exception handlers" just turns off luma's arm11 exception handlers. It doesn't prevent the crash from happening, it just lets the HOS handle the exception instead of using luma's exception handler which displays info on the crash.

It is also irrelevant since this is the errdisp applet, not an arm11 exception.

xerxesmanflayer, 2019-10-21:
Ah, thanks for the clarification.

Oreo639, 2019-10-24:
Can someone try reproducing the error and then paste their final entry of `/luma/errdisp.txt`

brunocar, 2019-10-24:
for sure, i'll edit this comment once i can do it

brunocar, 2019-10-25:
Edited! its quite long after experimenting with it

Oreo639, 2019-10-25:
Tysm.

brunocar, 2019-10-25:
you are welcome, anything you can tell us from it? i think its plenty obvious that errdisp is the the key to understanding this, but thats as far as my knowledge goes

Oreo639, 2019-10-25:
BOSS sends a Logged error.
I don't think this is supposed to be fatal, but since the problem explained here: https://github.com/AuroraWright/Luma3DS/pull/1332
the `info->type != ERRF_ERRTYPE_LOGGED` returns true causing it to be displayed as a fatal error.

Correct me if I am wrong.

Shadow322, 2019-11-05:
game works now on latest Luma3DS update as of today

brunocar, 2019-11-05:
Awesome! thanks for the help

xerxesmanflayer, 2019-11-05:
> game works now on latest Luma3DS update as of today

I dont see a new release on the updater, which version are you referring to?

Shadow322, 2019-11-05:
Latest comment get it at

https://github.com/hax0kartik/luma-hourlies/releases/tag/234-luma3ds-dfd50d9

xerxesmanflayer, 2019-11-05:
> Latest comment get it at
> 
> https://github.com/hax0kartik/luma-hourlies/releases/tag/234-luma3ds-dfd50d9

Thank you so much, I've been waiting to play this with my fiance for months <3

brunocar, 2019-11-05:
I tried it, it works.

🦀Errdisp is GONE🦀

xerxesmanflayer, 2019-11-05:
I can also verify, it's working for me as well

closed 2020-04-17 by TuxSH