LumaTeam/Luma3DS #1294: [Feature Request] 30 minute and 45 minute UTC deviation support in the NTP client

Shawnjoseph2001, 2019-07-17:
Currently, the NTP client is unusable for Luma3DS users in parts of French Polynesia, Australia and Canada and all of Iran, Afghanistan, India, Sri Lanka, Nepal, Myanmar, Lord Howe Island and New Zealand because the time zones the respective places above are located in are half hour or 45-minute deviations from UTC. For example, India is in the time zone UTC+0530. In the current implementation of the NTP client present in Luma3DS, only full hour deviations from UTC are supported, preventing Luma3DS users in India from being able to correctly set the time using the NTP client. While a minority of Luma3DS (and 3DS) users are in these locations and proper NTP client support is not an essential feature, proper NTP support for these regions would be appreciated by people living in these areas and travelers to or from those respective time zones.

TuxSH, 2019-07-22:
Will implement before next release.

TuxSH, 2020-04-17:
Someone else did

closed 2020-04-17 by TuxSH