LumaTeam/Luma3DS #1295: Rosalina Menu Freeze

neonoafs, 2019-07-18:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**11.10.0-43**

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**10.0.1**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (1)

Splash: (1 )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

Rosalina freeze and console stop working after pressing L+Down+SELECT.
Cant move the cursor. Have to force powering off the device

[VideoExample.zip](https://github.com/AuroraWright/Luma3DS/files/3404502/VideoExample.zip)

**Steps to reproduce:**

1.Turn device on

2. press L+Down+SELECT

**Dump file:**

massijay, 2019-07-27:
I have the same problem, with latest firmware (11.10.0-43) and the latest luma release (10.0.1), it freezes also during the first configuration when I press start to save

massijay, 2019-08-19:
I discovered that the X button was stuck always pressed, after fixing it Luma worked correctly

rossosat, 2019-12-09:
I have the same issue but in my case there is no buttons broken

neonoafs, 2019-12-09:
I solved the problem by changing my SD card.

closed 2019-12-09 by neonoafs