LumaTeam/Luma3DS #1297: Cheats from Boot NTR Selector doesn't work anymore

Axiskhan, 2019-07-20:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

Shadow322, 2019-07-20:
fill out the damn info

TuxSH, 2019-07-22:
luma 10.x removes support for it

closed 2019-07-22 by TuxSH

pthfdr-42, 2019-07-25:
> luma 10.x removes support for it

Then you should implement a way to *manually* search/compare/change decimal values in RAM like what Gateway used to offer. Sometimes preloaded cheat codes are not enough (and nonexistant for some games).