LumaTeam/Luma3DS #1300: Use *.pool.ntp.org instead of Google servers.

pthfdr-42, 2019-07-25:
`*.google.com` is blocked in China, and using a VPN is impossible on a 3DS.

TuxSH, 2019-07-31:
Heh right.

Do you have a better NTP server which IP doesn't change?

pthfdr-42, 2019-08-01:
Just let the user set the IP.

ParzivalWolfram, 2019-08-23:
This is a dumb suggestion, but is the Windows NTP server blocked in China?

Also, @TuxSH couldn't you check DNS to get the IP from hostname?

pthfdr-42, 2019-08-23:
`time.windows.com` is not blocked.
I think that yoh should just let the user to specify an IP.

danielhoherd, 2019-09-08:
FWIW [NIST runs a bunch of NTP servers with static addresses](https://tf.nist.gov/tf-cgi/servers.cgi), but they also give this advice:

> Whether you connect to a server using the name or the IP address, it is a bad practice to “hard-code” a particular server name or address into a device so that these parameters cannot be changed by the end user if that becomes necessary at some future time.

DMSalesman, 2019-09-09:
Luma has a configuration file, `config.bin`, but its contents aren't in plaintext. Could an user-provided URL be stored in there nevertheless?

pthfdr-42, 2019-10-22:
I searched SO and found [this answer](https://superuser.com/questions/307158/how-to-use-ntpdate-behind-a-proxy).  
You can use `curl` to access a server and extract the current time in the header.

    curl -H'Cache-Control:no-cache' -sI example.org|grep ^Date
Of course, I replaced Google with the "example" site since I do not think that it will be usually blocked.  
It is not very accurate but it should be accurate within 3-5 seconds, which is enough for a game console.

ParzivalWolfram, 2019-10-23:
I'd be willing to let you use my private site for this, if need be. (It's a
private file share for minor things, so it's forever under construction lol)

tomjschwanke, 2019-10-27:
Private websites are a bad idea since no one knows when you might pull the plug or how much load it can handle. Google has lots of datacenters to ensure high availability under high load.

I think the default should be one of the NTP pools but with the option to change it.

pthfdr-42, 2019-10-28:
@tomjschwanke I think that it should be "example.org" (it is a real site).  
It does not have any meaningful content so it should not be blocked.
Also it is managed by IANA itself.

ParzivalWolfram, 2019-10-28:
I'd also like to point out that if we assume the set/given DNS is correct (it should be or we couldn't connect to anything outside the local network at worst) we could just ask for a country to use and then ask the DNS for an IP so this static-IP thing is a non-issue for the most part?

tomjschwanke, 2019-10-29:
And maybe try the gateway first. Some routers provide NTP services 🤔

ParzivalWolfram, 2019-10-29:
usually not great though... it'd work as a fallback though.

TuxSH, 2020-04-17:
DNS resolution is not done because it takes too much code space (which is very limitedà.

TuxSH, 2020-04-17:
Replaced it with the IP of time.windows.com (by default)

closed 2020-04-17 by TuxSH

eku, 2020-04-18:
Could we please use services that respect privacy? Neither Google nor Microsoft do. Please reconsider using ntp.org.

ParzivalWolfram, 2020-04-18:
I mean, this is better than nothing and isn't impeded in most countries,
plus why would they log who requests *time?* That's not usable for
advertising or profit-making, so...?

DraconicNEO, 2023-04-06:
> I mean, this is better than nothing and isn't impeded in most countries, plus why would they log who requests *time?* That's not usable for advertising or profit-making, so...?
> […](#)

Though NTP.org is usually more reliable than the Windows NTP which many people have complained about and still complain about due to being inaccurate or going down randomly.