LumaTeam/Luma3DS #1302: [Compilation] Luma 10.x+ bugs and incompatibilities often encountered on NH

suppai, 2019-07-30:
This is a compilation of issues commonly encountered by Nintendo Homebrew Helpers with Luma 10.x, the latest version of which is 10.0.1 at time of writing. None of these issues occur on Luma3DS 9.1.

1. **ErrDisp codes are incorrect:** All ErrDisps seem to display "error code" 0x000000d1, 0x000000d2, 0x00000010, etc. which are actually "Success" according to Kurisu and Robocop-NG's error code database. Example:
```
An error occurred (ErrDisp)
Error type: generic
Process ID: 67115840
Address: 0x00000d1a
Error Code: 0x000000d1
Press any button to reboot.
```
1. **Erroneous ErrDisp in some(?) games:** The game "Heroes of Ruin" display an erroneous  0x000000d2 ("Success") error code, preventing their usage. [Here's a picture of the error.](https://media.discordapp.net/attachments/441119928334942218/605842349712605205/61380299-8b8e6e80-a8c2-11e9-9ae1-b66c68cd7d71.png). Similar but not as consistently reproducible issues have been reported with "New Super Mario Bros 2" and "Super Mario Maker".
1. **GodMode9 fails to init screens correctly when used as the START firm on consoles with IPS screens:** If `GodMode9.firm` is the only firm in `/luma/payloads` and the user has a console with at least one IPS display panel, GodMode9 is unable to init the screens when starting it by holding `START` on boot. Booting GodMode9 with b9s as `boot.firm` works, as does loading GodMode9 from the Luma payload chainloader menu when there is more than one `.firm` in `/luma/payloads`. Occurs only with GodMode9 v1.8.0, v1.7.1 works. However, v1.8.0 works in every other configuration besides this (b9s, fb3ds, as boot0/1, etc.), so I assume this is likely a Luma issue (but may be wrong).
1. **NTP sync feature unusable on CHN-region systems:** Mainland China blocks most Google servers, including their NTP servers, breaking this feature.
1. **Incompatible with 4.x.x:** Luma 10.x doesn't boot if the 3DS firmware is any 4.x.x version. A somewhat niche issue, but NH still encounters enough 4.x.x users to make it relevant.
1. **Out of memory crash with Old 3DS extended memory games:** Occasionally, pressing Home when in an Old 3DS extended memory causes the system to run out of memory, crashing it.

SirNapkin1334, 2019-07-31:
Just dropping this here for number 2...
https://github.com/d0k3/GodMode9/issues/466

cosmenp, 2019-09-06:
I have this problem with super smash bros too

nachoidv, 2019-09-07:
I got the same, with smash too, ir happens when i press the home button

Blue20006, 2019-09-12:
So what do I have to do to fix this error?

Blue20006, 2019-09-12:
Please i need help

Shadow322, 2019-09-12:
@Blue20006 no

Blue20006, 2019-09-12:
When the error occurs it puts me

An error occurred (ErrDisp)

Error type: generic
Process ID: 67115840
Adress: 0x00000d1a
Error code: 0x00000d1

Press any button to reboot.

Shadow322, 2019-09-12:
@blue20006 I said no

Jelliott64, 2019-09-16:
I'm getting a similar error on Mario Tennis Open when using a purple charge shot as Baby Mario

neox32, 2019-09-22:
hello I also have this problem but it appears to me when I turn on the console does not even appear the home menu I just turn it on and pum this error, I have luma 10.0.1 in an old
3ds xl someone could tell me how to fix it is that I don't know about this issue, so thanks in advance

brunocar, 2019-10-05:
oh so thats the bug, errordisp is the wrong code, anyone has a hotfix yet?

xerxesmanflayer, 2019-10-21:
Any news on a fix for Heroes of Ruin?

mokamota, 2019-11-16:
Got the same problem 😢

xerxesmanflayer, 2019-11-16:
https://github.com/hax0kartik/luma-hourlies/releases/tag/234-luma3ds-dfd50d9
Try this, it worked for me.

mokamota, 2019-11-16:
Sorry im new at this and i bought my n3ds xl already hacked so where do i need to put this

xerxesmanflayer, 2019-11-16:
> Sorry im new at this and i bought my n3ds xl already hacked so where do i need to put this

"The file you want to download is Luma3DSvX.X(.X).7z, not Source code (zip) or Source code (tar.gz)!"

"Extract boot.firm from the Luma3DSvX.X(.X).7z archive with a tool like 7-zip (Windows/Linux) or Unarchiver (macOS) to the root of your 3DS' SD card."

https://github.com/AuroraWright/Luma3DS/wiki/Installation-and-upgrade

xerxesmanflayer, 2019-11-16:
> Sorry im new at this and i bought my n3ds xl already hacked so where do i need to put this

On another note, welcome to the modding community ^_^

nachoidv, 2019-11-16:
> https://github.com/hax0kartik/luma-hourlies/releases/tag/234-luma3ds-dfd50d9
> Try this, it worked for me.

Thanks man :D it worked for me too

luck1804, 2019-11-25:
Thank you ... It worked for me too.

Pearl8520, 2020-02-23:
Getting this error with Pokemon alpha sapphire

LeanneMTL, 2020-02-29:
> https://github.com/hax0kartik/luma-hourlies/releases/tag/234-luma3ds-dfd50d9
> Try this, it worked for me.

May I ask what to do after downloading the file?

Oreo639, 2020-02-29:
Place the boot.firm on the root of your sd card.

TheOneEyedGrimReaper, 2020-04-14:
hello.
i'm on the latest eur firmware.
there is my problem:

An error occurred (ErrDisp)
Error type: invalid
Process ID: 1
Process name: loader
Process title ID: 0x0000000000000000
Address: 0x00000000
Error code: 0x00000010

this is happened when i'm tried to install a game from fbi.
the fbi is failed to install the game and when i'm pressed the home button just show this.

someone have any idea?

Shadow322, 2020-04-14:
> hello.
> i'm on the latest eur firmware.
> there is my problem:
>
> An error occurred (ErrDisp)
> Error type: invalid
> Process ID: 1
> Process name: loader
> Process title ID: 0x0000000000000000
> Address: [...]

We don't support piracy so take it elsewhere

xerxesmanflayer, 2020-04-22:
> We don't support piracy so take it elsewhere

Installing via FBI doesn't automatically mean he's talking about piracy, you can backup games you own into that format.

TheOneEyedGrimReaper, 2020-04-22:
> We don't support piracy so take it elsewhere

i don't know what's your problem guy,
i'm installing converted gba games through fbi.
and i love to know what is the game like before i buy it on cartridge.
i sucked enough with my ps2 in the old days because i'm not able the try out the games i buyed from the shop and i had to decide based on the cover.
i don't want wasting money anymore on games i don't like.
but if i assuming i am a madafaking pirate. this is none of your business.

the problem is solved easily.
the memorycard i had is a fake one and it's corrupted the files on it.
it's a miracle i haven't bricked my console when installing luma cfw with this card.
i brought a kingston one and placed all the files i need and it's work fine.
don't buy platinet memory cards from chinese shops!
buy a kingston instead!

xerxesmanflayer, 2020-04-23:
> i don't know what's your problem guy,
> i'm installing converted gba games through fbi.
> and i love to know what is the game like before i buy it on cartridge.
> i sucked enough with my ps2 in the old [...]

Oh, so you're confirming that it was piracy, ok then. Yeah, that was a stupid thing to admit to. I have nothing more to say to you.

J25Esguerra, 2020-05-10:
Hello there, i am a newbie in this modding i have the issue as well, i put my boot.firm on the root sd card i replaced my sd card as well. I couldn't figure out the answer with this my luma is 10.1.3-3143e7e

Sanaki, 2020-08-05:
For what it's worth, closing and opening the lid will init the screens for direct-booted godmode9. It's not optimal, but it does bypass the issue.

Shadow322, 2020-08-05:
> For what it's worth, closing and opening the lid will init the screens for direct-booted godmode9. It's not optimal, but it does bypass the issue.

I'll say it again just compile latest source as it's been fixed

TuxSH, 2020-12-14:
Only remaining issue is a GM9 issue, closing.

closed 2020-12-14 by TuxSH