LumaTeam/Luma3DS #1304: Super Smash Bros. for 3DS 1.0 (EU) (Gamecard) crash

EnterpriseFreak, 2019-08-02:
**System model:**
New 3DS

**SysNAND version (+emu/redNAND version if applicable):**
11.10.0-43E

**Entrypoint (How/what you're using to boot Luma3DS):**
Luma 10.0.1 as FIRM1/FIRM2 (installed via godmode9)

**Luma3DS version:**
10.0.1 and 9.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (After payloads)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Off)

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

The loader process crashes with an undefined instruction error on ARM11 core 1.
(If the update is installed the game will start but lockup on the loading screen after you press start on the title)

**Steps to reproduce:**
1. Launch the game

**Dump file:**
[crash_dump_00000002.zip](https://github.com/AuroraWright/Luma3DS/files/3463104/crash_dump_00000002.zip)

EnterpriseFreak, 2019-08-03:
The same error occurs on the old3DS too.
[old3ds crash dump.zip](https://github.com/AuroraWright/Luma3DS/files/3463542/old3ds.crash.dump.zip)

SamHep0803, 2019-08-03:
Can confirm on v1 cart.

TuxSH, 2019-08-07:
Error code is 0xd9004587 which is basically "corrupt contents".

closed 2019-08-07 by EnterpriseFreak