LumaTeam/Luma3DS #1311: [Feature request] modules for Rosalina menu

editedreality, 2019-08-29:
Okay so something i think that could be cool to add to Rosalina is modules, Modules is some file which you put on your SD card and if Rosalina detects files in that directory it adds a menu called Modules inside of that menu it will have all of the modules which can be menus or features. For example you could have a brightness module, you put it inside of the modules folder and open the modules menu inside of Rosalina and then click Brightness and inside of that are options for you to chose from.
This would be useful because people could add features to Luma without needing to download, edit and build the source code and then have people download the firmware.

Shadow322, 2019-08-29:
Brightness was killed off in like fw 9.4 that's why it's not in BootNTR anymore

Dimensional, 2019-08-29:
I think he was just using brightness as an example to explain the use of dynamic modules. Sounds something akin to Linux Kernel Modules.

Ty-Dye, 2019-09-02:
So kind of like things to add on to the luma menu? I like your idea though it may be really hard to code.

TuxSH, 2020-04-17:
No; too difficult

closed 2020-04-17 by TuxSH