LumaTeam/Luma3DS #1312: [Request] Enable Gateway

MaxLempira, 2019-08-29:
I'm one of those who is still using Gateway.
Would it be able to enable Gateway to be used on emuNAND, or patch System File instead 
 before Gateway emuNAND launches?

These are the files that prevents Gateway emuNAND to be launched:

Nintendo 3DS Gateway
0004003000008F02	        Home Menu
0004013000001C02	GSP System-Module
0004013000002D02	NWM System-Module
0004013000008002	        NS System-Module
0004013800000002 	NATIVE_FIRM

New Nintendo 3DS Gateway
0004003000008F02 	Home Menu
0004013000002D02	NWM System-Module
0004013020001C02	New3DS GSP System-Module
0004013820000002 	New3DS NATIVE_FIRM

Hope Luma can do something about this...

urherenow, 2019-08-30:
Gateway has nothing to do with Luma, and already uses emunand. It’s just not compatible above few 11.2. You can update your sysnand completely, and still have an 11.2 emunand for gateway use, no problem.

You will NEVER be in Gateway mode above 11.2. PERIOD.

fincs, 2019-08-30:
Piracy is not supported.

closed 2019-08-30 by fincs

urherenow, 2019-08-30:
@fincs although I agree with closin this (not that my approval maters), your reason is super lame. GW hasn’t been compatible with the latest FW for an extremely long time, and NEVER WILL. Luma is therefore much more suited for piracy. There is only one reason to use a GW now, and that’s because people prefer to use its cheat engine.