LumaTeam/Luma3DS #1314: Graphic Translations (in enable game patching mode) aren't patching

Hareon, 2019-09-02:
Hi guys!!!
I am part of a team to translate 3DS Pokémon Games to brazilian portuguese.
When I use a Xdelta program to patch the translation a rom and play on Citra, all works perfectly.
But I want to launch the translation for 3DS people to use original game cartridges too.

With Enable Game Patching option, some graphics works when I put'em on the folders, but, I don't know why, the Pokédex graphics don't work!

On Citra the graphics was there, but on 3DS the same graphics are showing in english, for some reasons I believe, Luma don't patch all the sectors of this game.
Someone know how to fix this issue?

I'm using Luma 10.0.1

The graphics on Citra are working, they are "Name of the Pokédex (Central kalos Pokédex, National Pokédex...) the type of Pokémon and the Search options.

The files are in the right folder following the original hierarchy.
[image]

In this print the right graphics are displayed on Citra.
[image]

Hareon, 2019-10-11:
Hi all! I have one more information about this problem.
Well, my team's partner tried to put the translated archives into .cia file and it works.

All graphics are working well when we put the translated graphics on the original rom and remake as .cia.

But I think this is a lot of work for people to apply the translation. Well, we are waiting for a possible fix to it in future. The Luma method to patch games are easy to apply and better. Thanks!

EtianAM, 2019-10-13:
Do you try to use Luma 9.1? I do not know much about Luma or how it works, but I can see that Luma 10.0 and 10.0.1 have some bugs in some games. Maybe that will solve it.

Hareon, 2019-10-21:
Hi! I am using the newer version, but I dont expect that the older version will fix it, because it's a translation problem and this can be a new problem for the developer.
In any case, I'll wait a position what I have to do.