LumaTeam/Luma3DS #1315: 3DS Luma Exception Error

ffff714, 2019-09-06:
**System model:**New 3DS

**SysNAND version (+emu/redNAND version if applicable):**11.10.0-43J SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**Boot9Strap

**Luma3DS version:**10.0.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
After I update 3DS from 11.9.0 to 11.10.0 by offline updater, some titles when launch will crash and report this.

[image]

Sorry, this is my fault but if can, help me please!

**Steps to reproduce:**

1.Offline update from 11.9.0 to 11.10.0

2.Launch specific titles (e.g. Monster Hunter Stories) and that will crash before load

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/3583893/crash_dump_00000000.zip)

Shadow322, 2019-09-06:
Update online

ffff714, 2019-09-06:
> Update online

yea but the result it's same...and I do some may solve this problem methods (reinstall title and trying ctrtransfer even format system) but that dosen't work too :(

closed 2020-04-17 by TuxSH