LumaTeam/Luma3DS #1319: Completely random crashes happen all the time

daemonspudguy, 2019-09-14:
New 3ds XL

11.10.43U

Sighax 

10.0.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4)

Splash: (before )

Splash duration: (1 )

PIN lock: ( off)

New 3DS CPU: (clock+l2 )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (x )

Enable game patching: ( x)

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: (x )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Seemingly random crashes happen. So many that if I tried upload all the dumps GitHub would explode

**Steps to reproduce:**

1.Do anything

daemonspudguy, 2019-09-14:
here are sample of crash dumps
[dumps.zip](https://github.com/AuroraWright/Luma3DS/files/3612877/dumps.zip)

EtianAM, 2019-09-14:
Do you try to desactivate the "New 3DS CPU: (clock+l2 )"?

daemonspudguy, 2019-09-14:
yes. I found my issue. Fastboot3ds seems to break things

closed 2019-09-14 by daemonspudguy

profi200, 2019-09-19:
I looked at many of these exception dumps and i'm sure fastboot3DS is not the cause of this. All except 2 happened in userland long after Horizon OS loaded properly. I would recommend reformatting your SD card. And also turn off the clock + L2 setting since it's known a few games don't like it.