LumaTeam/Luma3DS #1322: Monster Hunter Generation crashes on start up Luma3DS v10

hoangtrung1999, 2019-09-21:
**System model:**
New 2DS LL
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
Sorry I'm newbie so I dont know where to find this.
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
v10.0.1
[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: (Off )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
I downloaded the legit CIA from [https://www.ziperto.com/monster-hunter-generations-3ds-legit-cia/](url), it's installed successfully but when I start it, it just keep crashing. Here is the image I took when it starts up
[image]

**Steps to reproduce:**

1.

2.

**Dump file:**

I don't have my laptop nearby so I can't upload at this time, will add it later.

Oreo639, 2019-09-22:
> I downloaded the legit CIA from [REDACTED]

Please don't post piracy links here.

I don't think you are going to recieve help with a pirated game.
Can you see if you can reproduce this on an offically purchased copy of the game?

> Sorry I'm newbie so I dont know where to find this.

It's the system version (you check in system settings) of your sysnand or emunand if you are running it from an emunand.

Ammako, 2019-09-26:
This is on your end, and not a Luma issue.

closed 2020-04-16 by TuxSH