LumaTeam/Luma3DS #1324: Using cheats on Pokemon games crashes the 3ds

whitetigerx8, 2019-09-25:
Whenever I try to use any cheats on any of the Pokemon games (X,Y,OR,AS,S,M,US,UM) lately, my 3ds crashes and I have to power it off and on again to get it working. The cheats I use don't seem to matter as all of them cause the crash to happen, but I do stick to ones that are marked for the version of the game I have installed. These game are directly installed onto my 3ds, and aren't the cartridges.

I've attached a dump file it gives me when it crashes:
[crash_dump_00000006.zip](https://github.com/AuroraWright/Luma3DS/files/3654333/crash_dump_00000006.zip)

leo60228, 2019-09-26:
I'm pretty sure the Luma3DS team considers this to be a bug in NTR.

916253, 2019-10-03:
@leo60228 there was no indication that the user was using ntr

LonelyAnalysis, 2019-10-04:
Some steps to go through:
1) (Sanity Check) make sure the games load properly without crashes without cheats on both 9.1.and 10.0.1.
2) 
With 10.0.1, use Checkpoint to generate the cheats file for each game.
With 9.1, find stable NTR plugins for the games you want to enjoy.
3) Experiment with the cheats in each game.
Note the amount of time it takes to crash, will the cheat engine even accept the cheat, and do the cheats actually work.
4) For each, experiment with the cheats and see which ones cause the game to crash.  Important, make sure you are using cheats for the current patched version of your game.  This can be checked in multiple places, I usually just go into FBI and check what patch says the version is.

5) Note the results for each combination of the above.

I think you're SOL for now, but the Luma team is aware of it.  Current best recommendation is use Luma9.1 and make sure you're running good plugins.  No word on a fix will come out for Luma 10+.

KevinTheStarfish, 2019-12-08:
Um how do i get luma 9.1 i have 10.0.1 and it doesnt change to 9.1

closed 2020-04-17 by TuxSH