LumaTeam/Luma3DS #1326: Cheats not working

flutterbrony, 2019-09-29:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.6.0-39U SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

[Sys 11.11.0-43E]

**Luma3DS configuration/options:**

Screen brightness: (1)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Enable loading external FIRMs and modules: (x)

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )(i don't see this option ?)

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Hello, I don't know if I'm in the right place sorry, but I'll post my issue here, and I'll move it if needed.

I use Sharkive to download cheats for my 3ds games, but when I boot the game, the cheat is shown, I can activate it, but it'll do nothing in-game.

I checked the files, and in sd:/luma/titles/<GameTitleId>/, there is a cheat.txt files, with the cheats correctly formatted in it.

I also checked if game Patching is enabled.

Sometimes, and in some games, when I activate the cheat, the game crash, and I have to reboot my 3ds, meaning that maybe, the cheat activated is doing something ?

**Steps to reproduce:**

1. Open Sharkive and download some cheats for a game

2. Open the game, then, Press  the combo button to open Rosalina menu, then go to Cheats..., and select a cheat code to activate.

3. Quit rosalina, then try the cheat in-game

4. Nothing happen (even if rebooting the game)

5. If rebooting the game, the selected cheat will not be selected anymore

**Dump file:**

rctgamer3, 2019-09-29:
Sharkive is dead, its code was merged into Checkpoint. Try with Checkpoint first.

LonelyAnalysis, 2019-10-04:
With recent releases with Checkpoint, Sharkive was merged into Checkpoint and now no longer downloads cheats from the internet.  Cheats are now prepackaged into Checkpoint and the cheats file will be created on the fly.

Of note:
You can get rid of Sharkive, it will just soft quit when booted.
While in Checkpoint, titles recognized by Checkpoint will have options to save, but in the bottom screen there's a cheats button.  You can use this create a cheats file for your games.
Remember what version your games are on, and only try to select cheats for that version.
There is an existing bug with cheats being unstable on the most recent Luma, it is being worked on.  A current idea floating around is to use the previous release of Luma to allow use of NTR plugins again.

tldr: sharkive is dead, long live checkpoint

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai