LumaTeam/Luma3DS #1328: Luma3DS (v10.0.1) Not Working on 3DS 4.2.0-9U

DeadSkullzJr, 2019-10-01:
**System Information:**
-------------------------------------------------------------
**System model:**
Nintendo 3DS (Old)

**Firmware:**
4.2.0-9U (SysNAND), 11.10.0-43U (EmuNAND)

**Entrypoint** _(How/what you're using to boot Luma3DS)_**:**
Boot9Strap

**Luma3DS Version:**
v10.0.1 (Downgraded to 9.1, reason posted below)

**Luma3DS Configuration:**
-------------------------------------------------------------
**Default EmuNAND:**
1

**Screen brightness:**
1

**Splash:**
Off

**Splash Duration:**
1

**PIN Lock:**
Off

**Options Enabled:**
- Enable Game Patching
- Show NAND or user string in System Settings

**Issue:**
-------------------------------------------------------------
NOTE: 
Before reading onward to the issue, YES I know I am trying to run Luma3DS on an old firmware version of the 3DS, however this was for a setup that uses EmuNAND. I already have two systems with modern custom firmware setups already, I merely wanted to maintain my SysNAND firmware for a bit of nostalgia. Now that I mentioned that (I honestly don't need people sniping me down about what I do to **my consoles**), onward to the issue.

During the process of setting up custom firmware on the system, I noticed that I am unable to reach the home screen at all when using Luma3DS (v10.0.1), no errors pop up or anything of the sort, it just hangs on a lit black screen. I can still access the Luma3DS settings, GodMode9, and even my EmuNAND (which is running 11.10.0-43U at the moment). I decided to bump it down to Luma3DS (v9.1) and that seemed to fix that issue. Unsure what was changed between the two updates other than what I saw in the changelog (unless one of those changes messed it up somehow). Knowing that 4.2.0-9U seems to work fine with Luma3DS (v9.1) I am going to say there is a kink with the latest version. I would try 10.0.0 but I doubt that would provide any better results, plus that version is badly bugged based on the notes I saw about it. Thanks in advance for taking the time to read the issue!

Shadow322, 2019-10-01:
Update to 11.10/11 as Luma3DS has been long time minimum fw of 7.x no lower

DeadSkullzJr, 2019-10-01:
> 
> 
> Update to 11.10/11 as Luma3DS has been long time minimum fw of 7.x no lower

That's not correct, the changelogs for both Luma3DS 10.0.0 and 10.0.1 include nothing about 4.X.X not being supported anymore, if it works on Luma3DS 9.1 then obviously there is an issue with Luma3DS 10.0.X. Luma3DS 8.1 added support for 4.X.X, it would make zero sense to add support for an older firmware then all the sudden removing it.

Vresod, 2019-10-03:
Launch safe mode and exit via the cancel button. This might work? i dont know.

Shadow322, 2019-11-08:
there is no reason to be on 4.X.X

RocketRobz, 2019-11-08:
> there is no reason to be on 4.X.X

Doesn't matter. The issue still stands.

fox4u, 2020-03-24:
Same issue here. SysNand 4.1.0-6U (even older lol), EmuNand 11.1.0-34U. Got black screen when booting SysNand with Luma3DS 10.0 and 10.0.1, while 9.1 works fine. EmuNand works great with 9.1, 10.0, 10.0.1 though.

TuxSH, 2020-04-17:
Latest commits fix it.

closed 2020-04-17 by TuxSH

fox4u, 2020-04-18:
Confirmed SysNand boots fine with a build at commit 5c16836626cb2474142cd57ddb416259c8226242. Thanks!