LumaTeam/Luma3DS #1329: Crashing in randomized Pokemon Y

aAvocadont, 2019-10-03:
System Model:
New 2DS XL

Firmware:
11.10.0-43U, not sure about cfw

Entrypoint (How/what you're using to boot Luma3DS):
Boot9Strap

Luma3DS Version:
v10.0.1

Hey, all. 
I'm *very* new to modding, and recently downloaded cfw onto my 3DS with no issue. Wanted to do a randomized nuzlocke of Pokemon Y, so I used PK3DS to do it. At first all was good, but I game over'd so I rerandomized and booted the game up. All is good, I talk to the rivals, I get my starter Pokemon, but when I go to see what stats and nature my starter is, that's where we run into issues.
When accessing the 'Pokemon" menu to rearrange and view Pokemon, the screen goes black and prompts me with a crash report. Reports below.

http://www.mediafire.com/file/3l88w2nsu4mfu41/crash_dump_00000000.dmp/file

Like I said, I'm very new, so I have no idea what most of these terms mean and how to access what. Help is much appreciated. I'm not sure whether this is a Luma problem or a PK3DS problem, so let me know if I need to go to PK3DS's page.

AuroraWright, 2020-11-17:
Does this happen with latest?

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai