LumaTeam/Luma3DS #1330: What is star(*) at Cheat list?

Duo-S, 2019-10-03:
[image]

Like this picture, what does the star(*) on the right side of the description mean?

This is the cheat code of Pokemon X

LonelyAnalysis, 2019-10-04:
There is additional information with the cheat, press Y to display the usage description.

closed 2019-10-04 by fincs

AuditeMarlow, 2022-04-11:
I've got a bunch of those cheats but pressing Y does not provide me with extra information unfortunately. Was this feature scrapped or is this a new/recurring bug?

hurda, 2022-04-11:
> I've got a bunch of those cheats but pressing Y does not provide me with extra information unfortunately. Was this feature scrapped or is this a new/recurring bug?
Cheats with * require a key to be activated, note the descriptions.