LumaTeam/Luma3DS #1334: My Homebrew CIA Says It "Needs to be updated before it can be launched."

DremOSDeveloperTeam, 2019-10-25:
Hi
I'm getting an error when trying to run a new piece of Homebrew software I've been working on as CIA. It says that the "Software must be updated before it can be launched," similar to the error you get when trying to run Animal Crossing: New Leaf after the Amiibo update.
It only does this when changing the [CIABUILDER](https://github.com/Manurocker95/CIABUILDER/) key from `0xFAA5C` to `0xOTNMC` (or any other key). I've been searching around for a few days and haven't found anything that can help me. Any suggestions?

fincs, 2019-10-25:
There's no reason to run homebrew as a CIA. 3dsx applications launched through Rosalina's native 3dsx support have full system permissions.

fincs, 2019-10-25:
If the homebrew menu takes 5 minutes to load, then it might be a good idea to clean up your /3ds folder, and/or start moving old homebrew apps to subfolders (which can be navigated).

3dsx in general is more compatible with entrypoints. Users who don't have Luma3DS can use the old hax 2.x system; which still works on a vanilla 3DS system.

profi200, 2019-10-27:
Maybe organize all the .3dsx homebrews in subfolders? That should reduce loading times by a lot.

As for the described problem: Sounds like a conflict with other titles installed on your system. Or the title/ticket.db is corrupt somehow.
Conflicts can happen because there is no standard and everyone takes whatever title ID and product code they want.

DremOSDeveloperTeam, 2019-10-27:
Perhaps, but every title ID I try doesn't work unless I usw the default CIABUILDER one. I can try some different title IDs later though.

profi200, 2019-10-30:
Unless you need very special permissions i would also recommend using .3dsx. Leaves no traces on the system as a bonus.

closed 2020-04-25 by TuxSH