LumaTeam/Luma3DS #1335: Possible Cheat Engine Bug

jkwaak1980, 2019-10-27:
**System model:**
New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.10.0-43U

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap/Sighax

**Luma3DS version:**
10.0.1

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (off)

Splash duration: (1)

PIN lock: (off)

New 3DS CPU: (off)

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

**Explanation of the issue:**
The cheat engine crashes my 2DSXL anytime I enable a cheat for any game.

**Steps to reproduce:**
1. Enable a cheat via the Rosalina menu

2. Re-Enter the game

3. Play a few seconds to minutes

4. Crash

**Dump file:**
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/3776329/crash_dump_00000000.zip)

Swearie, 2020-02-01:
I am also experiencing a lot of crashes when using rosalina cheat system. I hope this gets fixed

AuroraWright, 2020-11-17:
This was fixed

closed 2020-11-17 by AuroraWright