LumaTeam/Luma3DS #1337: Random crashes with Mario Kart 7

V10lator, 2019-10-29:
**System model:** New 2DS XL
**SysNAND version (+emu/redNAND version if applicable):** 11.11.0-43E
**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap
**Luma3DS version:** 10.0.1 stable
**Luma3DS configuration/options:**
Screen brightness: 4
Splash: Off
Splash duration: 1
PIN lock: Off
Enable loading external FIRMs and modules: ( )
Enable game patching: (x)
Show NAND or user string in System Settings: (x)
Show GBA boot screen in patched AGB_FIRM: (x)
Set developer UNITINFO: ( )
Disable ARM11 exception handlers: ( )

**Explanation of the issue:**
It crashes randomly. Image and dump are from the last crash which happened while exiting the game.
[image]

**Steps to reproduce:**
1. Play the game
2. Wait (might take a long time)

**Dump file:**
[crash_dump_00000004.zip](https://github.com/AuroraWright/Luma3DS/files/3784365/crash_dump_00000004.zip)

Shadow322, 2019-11-08:
Did u try latest build released 3 days ago

https://github.com/hax0kartik/luma-hourlies/releases/tag/234-luma3ds-dfd50d9

CyrilCommando, 2020-02-19:
Same problem. Been happening for a very long time.

[crash_dump_00000013.zip](https://github.com/AuroraWright/Luma3DS/files/4222972/crash_dump_00000013.zip)

CatsLover2006, 2020-05-24:
I have the same entry point and a similar version, New 3DS XL, but I don’t have this issue. Are you using a cartridge or digital copy?

V10lator, 2020-05-25:
I'm using a digital copy (preinstalled on the console / no warez).

Shadow322, 2020-05-25:
Why have game patching enabled?

CatsLover2006, 2020-05-26:
By the appearance of the crash, looks like you have the same issue I’m having while making a custom track: one of the files (for me it’s the .bclgt file) are improperly formatted, causing the 3ds to crash. You might need to re-install the game.

CatsLover2006, 2020-05-26:
**Edit:** I say this from experience. I’ve had corrupted files two times: one in the situation described above, one where my cartridge wasn’t making 100% contact and files weren’t getting read properly. This was before I hacked my (old, at the time) 3ds. Over time, things in the game disappeared or glitches until the game closed with an error, forcing a restart. Corrupted files are likely your issue, speaking from experience.

CatsLover2006, 2020-05-26:
> Why have game patching enabled?

For reference, CTGP-7 needs it enabled. I use it all the time on my New 3DS XL with no issues.

Shadow322, 2020-05-26:
> 
> 
> > Why have game patching enabled?
> 
> For reference, CTGP-7 needs it enabled. I use it all the time on my New 3DS XL with no issues.

well first u didnt say u was using a hack as CTGP-7 isnt Mario Kart 7

CatsLover2006, 2020-05-29:
> > > Why have game patching enabled?
> > 
> > 
> > For reference, CTGP-7 needs it enabled. I use it all the time on my New 3DS XL with no issues.
> 
> well first u didnt say u was using a hack as CTGP-7 isnt Mario Kart 7

CTGP-7 is MK7 it just patches game files. Many of them, considering all the features it adds.

**Note:** Game patching shouldn’t do anything if there’s nothing in the game-patched file folder.

AuroraWright, 2020-11-17:
Does it happen with game patching disabled?

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai