LumaTeam/Luma3DS #1341: It's  possible launch on startup a custom menu?

tetradog, 2019-11-23:
I wonder to develop a menu style switch to launch games,apps and config 3ds but i don't know if it's possible...

The most important is the home menu replace it to the custom...i think it's possible because since v8 can L+Select+(down?) Then home button can be override with the new custom menu no?

[Sorry my bad English]

Shadow322, 2019-11-23:
This is not a Luma3DS issue take this to gbatemp forums

tetradog, 2019-11-23:
:( but Luma is the software necessary to init program on startup and edit home button...no?

Multimegamander, 2019-11-28:
It's possible, other CFWs have a patch to load a different tid than home menu, the Luma devs said they don't want to add it tho

Multimegamander, 2019-11-29:
https://github.com/Multimegamander/ProjectAtlas

Me, RocketRobz (creator of TWiLightMenu++) and TheGuy3DS are making a 3DS Home Reimplementation.

profi200, 2019-12-03:
What the hell is MultiCTR?

Multimegamander, 2019-12-04:
MultiCTR is a new CFW we're making in C++ that takes features from Corbenik, Luma, and Cakes and adds some unique features.

RS55, 2019-12-23:
@Multimegamander I hope that, if it becomes a reality, you will do everything Luma3DS left out of annoyance or disinterest (and are logically applicable but have not done so, such as letting you select which screen to take screenshot and assign a combination of buttons or a button to take screenshot).

RS55, 2019-12-23:
being frank, I hope that, if it becomes a reality, you will do everything Luma3DS left out (and that are logically applicable but have not done so, such as letting you select which screen to take screenshot and assign a combination of buttons or a button to take screenshot ...).

closed 2020-04-15 by TuxSH