LumaTeam/Luma3DS #1343: Pokemon Crystal crashes on startup, latest Luma, original VC title and injected.

Nashismo, 2019-12-06:
-- Please fill in the placeholders.-->
**System model:**

Old 2DS

**SysNAND version (+emu/redNAND version if applicable):**

11.13.0-45U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v10.0.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (3 )

Splash: (off )

Splash duration: (1 )

PIN lock: (off)

Enable loading external FIRMs and modules: ( off)

Enable game patching: ( off)

Show NAND or user string in System Settings: (off )

Show GBA boot screen in patched AGB_FIRM: (off )

Disable ARM11 exception handlers: (off )

**Explanation of the issue:**

Pokemon Crystal (GBC) crashes, I hace used the official VC game and it does not work.

I will leave the dump file.
[pokemon_crystal_usa_official_vc(dump).zip](https://github.com/AuroraWright/Luma3DS/files/3929950/pokemon_crystal_usa_official_vc.dump.zip)

Thank you very much.

Multimegamander, 2019-12-08:
try fixing it through the EShop

Nashismo, 2019-12-08:
> 
> 
> try fixing it through the EShop

What do you mean? Thank you.

RocketRobz, 2019-12-08:
> What do you mean? Thank you.

Redownload in the Downloads list.

Nashismo, 2019-12-08:
> 
> 
> > What do you mean? Thank you.
> 
> Redownload in the Downloads/Purchases list.

Oh no you see, this is a CIA that I downloaded from the net, it is an official version, not an injected ROM. But I did NOT downloaded the game from my own Eshop.

Maybe this is a problem because this game has a wireless link feature? Maybe this game is simply impossible to install unless you purchase the game as intended, if that is the case, then I guess I'll buy it. I just want to know what the problem is.

fincs, 2019-12-08:
> this is a CIA that I downloaded from the net

Or in other words, it's a pirated copy which might or might not be corrupted. Piracy is not supported.

closed 2019-12-08 by fincs

Nashismo, 2019-12-09:
Piracy is not supported, said the people who hacked the 3DS wide open, mostly for piracy... hahahhhaha fine.

fincs, 2019-12-09:
We did not hack the 3DS wide open for piracy. We hacked it for homebrew. Pirates took our work and ran away with it.

Nashismo, 2019-12-09:
Ok, it is fine, I was ignorant of this.