LumaTeam/Luma3DS #1344: NTR is not working for some people

Tibbyman, 2019-12-07:
Yesterday I tried using NTR to record for the first time. This is what it said "Loading 3.2/3.3/3.6 ...
Load failed !
#CUSTOM_PM_PATCH_FAIL
Press any key to reboot"
 I have no idea what the problem might be. I asked someone at sthetix's discord server and they said that "something about how the issue was they were using the plugin loader version of luma". I had no idea what that meant. ANother person from the same server also said they had the same issue. We were both running Luma3DS v10.0.01. Does anyone know how to fix this. I'm trying to learn how to record gameplay on my 3ds.

fincs, 2019-12-08:
NTR is not supported as of Luma3DS 10.x due to incompatibilities with Luma's PM reimplementation (among other things).

> Note: the pm reimplementation currently breaks NTR CFW to some extent (in particular game plugins). The abandoned "NTR CFW" is now unsupported, even if it may work.

closed 2019-12-08 by fincs

Ammako, 2019-12-09:
This is not the problem, the problem is that the custom fork of Luma they are using causes incompatibilities (BootNTRSelector was updated recently to work with Luma 10.)

Either way, this is not a Luma issue, you'll have to take it up with whoever made that fork.