LumaTeam/Luma3DS #1345: Can't Download a File on Luma3DS

niffy603, 2019-12-08:
I was looking in the Luma3DS files, but can only find boot.firm and boot.3dsx in the file. I downloaded the latest version and all 3 assets and it wasn't in any of them. 
If anyone could help me find where to download boot.nds, it would be of great help.

Oreo639, 2019-12-08:
> If anyone could help me find where to download boot.nds

What are you doing that would require a boot.nds file?

niffy603, 2019-12-08:
I'm trying to install Boot9strap

Oreo639, 2019-12-08:
> Copy boot.nds (B9STool) to the root of your SD card

It clearly says where to find it.

niffy603, 2019-12-08:
I've been using that guide. When I download the file, it's just not there. 
[image]

Oreo639, 2019-12-08:
boot.nds is not apart of luma

niffy603, 2019-12-08:
Oh, I see now. Just reread it. Thanks.

Oreo639, 2019-12-09:
Just a reminder. If the issue is resolved, it would be best to close it.

closed 2020-04-15 by TuxSH