LumaTeam/Luma3DS #1346: Game crashes with ErrDisp error

spnshguy, 2019-12-10:
System model: New 3DS XL
Entrypoint (How/what you're using to boot Luma3DS): Boot9Strap
Luma3DS version: 10.0.1 stable

While playing Luigi's Mansion, the game always crashes while playing the game (opening a door to be more specific).

Let me know if more details is needed

`An error occurred (ErrDisp)
Error type: generic
Process ID: 67115840
Address: 0x00000d0a
Error Code: 0x000000d0
Press any button to reboot.`

Multimegamander, 2019-12-10:
Open EShop, and redownload in the Downloads/Purchases list.

If it's a pirated game, we won't help you.

spnshguy, 2019-12-10:
I tried that but it still crashes :/

Oreo639, 2019-12-10:
Try the latest Luma hourly. Also, it would be helpfull if you would share your /luma/errdisp.txt (the latest release doesn't display the correct information)

spnshguy, 2019-12-10:
Here is the error log that you asked for

```An error occurred (ErrDisp)
Error type: generic
Process ID: 39
Process Name: LM3DS
Address: 0x0010f9d8
Error Code: 0xd900458b
Press any button to reboot.```

[errdisp.txt](https://github.com/AuroraWright/Luma3DS/files/3947416/errdisp.txt)

spnshguy, 2019-12-12:
@Oreo639 let me know if you need any more info

TurdPooCharger, 2019-12-27:
https://github.com/hax0kartik/luma-hourlies/releases

DMSalesman, 2020-01-08:
I didn't grab a log, but a similar crash out of thin air occurred to me with Fire Emblem Awakening (EUR), on an Old3DS with the latest Luma version.

TuxSH, 2020-04-15:
Latest commits/oreo's PR should have fixed it.

closed 2020-04-15 by TuxSH