LumaTeam/Luma3DS #1347: Boot NTR CFW SELECTOR not working

GalaxyGold, 2019-12-17:
https://github.com/44670/BootNTR/releases
This link is NTR CFW downlad link. 

My 3ds is boot9strap CFW. 
recently, Nintendo release 11.13 ver update. 
necessarily, I updated my 3ds today. 'cause I can't using friend add function. 
Everyday I always using NTR CFW app. 
but after the update, NTR CFW not working. :( 
NTR CFW app said 'Loading 3.6...
Load failed !
#firmware unknown
##Delected firm :2.56
Press any key to reboot'

Please help. 
What can I do? 
I want to use NTR CFW app.

Oreo639, 2019-12-17:
That is BootNTR. Not BootNTRSelector.
This is probably what you are thinking of:
https://github.com/Nanquitas/BootNTR

GalaxyGold, 2019-12-17:
I solved myself. 
I installed 'BootNTRSelector-PabloMK7-Banner.cia'. 
link: https://github.com/Nanquitas/BootNTR/releases

and I clicked 3.6. 

Thank you anyway 'Oreo639'.

closed 2019-12-17 by GalaxyGold

miiforlong, 2023-06-07:
it doesn't work in 2023 .......................