LumaTeam/Luma3DS #1351: [Feature Request] Option turn off bottom screen

40VeR, 2019-12-23:
Please add the option to turn off the bottom screen with a button combination. This would make playing emulators in dark environments possible for me due too my eye condition and maybe more pleasant for some people plus it saves batterylife.

Shadow322, 2019-12-23:
LOL no

RocketRobz, 2019-12-23:
Retroarch allows turning off the bottom screen by touching one vertical half of the touch screen.

RS55, 2019-12-23:
Well, being frank there is a lot of game where it could be applied and not affect the gameplay, besides the idea would be to turn off the light, not turn off the screen, for battery saving it would be remarkable

40VeR, 2019-12-23:
> 
> 
> Retroarch allows turning off the bottom screen by touching one vertical half of the touch screen.

This is indeed true but sadly the cores in retroarch don't perform well on the 3ds like the ported emulators by bubble2k16 or mgba by endrift (Whose core has been removed from retroarch)

> 
> 
> Well, being frank there is a lot of game where it could be applied and not affect the gameplay, besides the idea would be to turn off the light, not turn off the screen, for battery saving it would be remarkable

Yes, I meant turning off the backlight, thanks for correcting me.

profi200, 2019-12-23:
This is not really Luma's job. You should ask the emu devs to implement this on their side.

40VeR, 2019-12-26:
> 
> 
> This is not really Luma's job. You should ask the emu devs to implement this on their side.

That's why it's a request. Also asking the devs is not really possible most emus haven't been touched in years.
And as @RS55 mentioned it could be used in licensed games.

TurdPooCharger, 2019-12-27:
Take black cardboard or thick cardstock paper and cut this to the dimension of your lower screen. Add [3M Scotch reusable sticky tabs](https://www.walmart.com/ip/3M-Scotch-Reusable-Mounting-Strips-1-x-3-6-Pkg/17209135). Put this on lower LCD when you wish to block it.

When you want to use the lower screen again, remove and mount the cardboard cut + sticky tabs to the backside of your upper LCD screen.

Eduxki, 2020-01-07:
So funny I think I'm gonna puke

closed 2020-04-15 by TuxSH